logisim-evolution/logisim-evolution #343: Small problem with the "Combinational analysis" window. V3.3

mbaillif, 2020-02-12:
Small problem with the "Combinational analysis" window.

The window is too narrow in some languages Example:de 
And buttons no longer appear spontaneously

maehne, 2020-02-12:
@mbaillif: Thanks for the bug report!

Could you provide a screenshot illustrating the issue?

What do you mean with "buttons no longer appear spontaneously"?

mbaillif, 2020-02-13:
After that, I think I'm a bit of a maniac. :-))

In fact that only happens in certain rather rare situations.
If you open the program in a language Example:en, open the "Combinational analysis" screen and then change the language Example:de screen does not resize.
[image]

and some buttons are missing

[image]

As I'm working a lot on the documentation at the moment I switch from one language to another without really realizing it and I observed this phenomenon without realizing that this case is rather extraordinary and rare.

BFH-ktt1, 2020-02-13:
I'm not sure exactly, but it could also be related to the Java Version. Note that the size of the window will not change when changing a language, only the "labels" on the button(s), Tabs, etc. will change; it is well possible that this might force some buttons to disappear; however, this is not likely related to logisim I think, but more to how java is handling labels being larger as the button/tab lying underneath.
Anyway, I agree completely with you that this is quite annoying.

maehne, 2026-08-21:
Closing as resolved by PR #2639. @mbaillif: Please reopen if the issue persists!

closed 2026-08-21 by maehne