logisim-evolution/logisim-evolution #345: Error when connecting register to itself

FamiliarBreakfast, 2020-02-13:
Yes, I know how stupid that sounds, but hear me out.

I was designing a bus for a computer model I was working on, and as part of the bus control, I had a single register for preserving the last non-zero value of the bus, if I ever needed it (I realize now I didn't need it, but whatever). The way I set it up, I had a system to check if the bus was a non-zero value. If so, it would bring the write-enable high causing it to read from the bus, and store its value. As soon as the bus became zero again, it would pull the write-enable low, preventing the register from storing 00.

The system worked fine until I attached the output to the bus. I wasn't stupid and added a control circuit to only output the value of the register when I sent the command to do so. However, as soon as a value was loaded into the bus (by external means) it immediately errored that line as well as everything else connected.

I'm afraid I don't have screenshots or files as I lost the setup while writing this, but I figured I'd mention it anyway incase someone else has this issue.

I think that maybe if you added some sort of anti-feedback mechanism, where the register wouldn't store the bus value while also outputting a value, it maybe would have fixed it. But that's only my head theory.

maehne, 2020-02-13:
Without any circuit file allowing us to reproduce your issue, it is hard to diagnose the cause of your problem. If the bus lines turned red, you have multiple sources driving your bus. In that case, you need to check your control circuits for your tristate buffers. If simulation stopped with an error message, you probably introduced inadvertently a combinatoric loop into your circuit. Such loops are created if you feedback some output of a combinatoric circuit to one or more of its inputs. This either will cause oscillation of the feedback signals or memorisation. Oscillation is sometimes exploited in ring oscillators and memorisation is exploited in latches. However, you should use the dedicated Logisim-evolution primitives for oscillators and latches instead of building your own using basic logic gates. Because the simulation of the latter is very fragile and could be only improved by giving Logisim-evolution the possibility to model gate delays. Issue #337 is somewhat related to this.

Your suggested solution of an anti-feedback mechanism is basically already in place. Logisim-evolution will give an error and halt simulation if signal values don't stabilise after a maximum number of delta cycles. Combinatoric loops can be prevented by inserting registers to break the loop to restore causality.

FamiliarBreakfast, 2020-02-13:
I can get you a test file in time

MarcinOrlowski, 2021-07-13:
Closing, as there's no OP follow-up. Please reopen if needed.

closed 2021-07-13 by MarcinOrlowski