logisim-evolution/logisim-evolution #347: I need a little help for doc

mbaillif, 2020-02-13:
I need a little help.

Can someone who develops in logisim confirm that the "JAR Libraries" part of the documentation is still up to date with the current state of the code.

Eventually, communicate corrections to me.

BFH-ktt1, 2020-02-13:
Unfortenately I cannot help in this issue, as this part of the code I never investigated/worked on, maybe @kevinawalsh has some more info here, as I remember he had some pushes in his holycross version related to "JAR Libraries"

kevinawalsh, 2020-02-14:
The text looks fine. The code examples are almost certainly out of date, as various APIs have changed over the years, but you can get a fresh copy from the src directory that is probably working. You can compile the gray code library to test it -- I believe that code still works fine.

There are two large gaps in the documentation though: instructions for building Jar libraries (specific versions of java are needed these days to avoid incompatibilities); and there is a new type of "Logisim Circuit" library that isn't mentioned at all in the documentation, I think.