logisim-evolution/logisim-evolution #352: exitAfterStartup and tty

klenin, 2020-02-22:
When running with `-tty table` option, Logisim hangs after displaying the table.
It can be trivially fixed by replacing `return` with `System.exit(0)` in `Startup.run`
after the call to `TtyInterface.run`.
However, I am quite confused about why this whole mess with `exitAfterStartup` 
and multiple direct exits from e.g. `parseArgs` is needed at all.
Can somebody please enlighten me?
For a quick fix, I can still provide one-liner patch of course.

BFH-ktt1, 2020-02-23:
I know that the -tty is still very broken, whilst merging the holycross additions I skipped some steps related to the headless execution of logisim. This is on my todo list, and as soon as I have again a bit time (currently I am very busy preparing two new courses) I will again look into this issue, please have a tad of patience

klenin, 2020-02-24:
> I am very busy preparing two new courses
Same :)
This semester I hope to upgrade my course from plain Logisim to Evolution.

Meanwhile then, please apply referenced PR -- it fixes the problem for now,
should be good enough until you have time for a proper refactoring.

maehne, 2020-02-27:
I guess with PR #354 being merged, this issue can be closed now, as there is still the issue #342 regarding additional problems with the `-tty` option.

klenin, 2020-02-27:
> I guess with PR #354 being merged, this issue can be closed now, as there is still the issue #342
> regarding additional problems with the `-tty` option.
IMHO it is the other way around.
Issue #342 is fully understood and fixed. Perhaps there is a slightly better way, but the difference is quite minor.
This issue, however, is patched up, but I really do not understand why the code is structured the way it is, and 90% sure it should be reorganized.
Multiple calls to `System.exit` from non top-level code make me uncomfortable :)

MarcinOrlowski, 2021-07-13:
While `Startup.java` is far from perfect, I think the issue reported is addressed to is seems like we can have this one closed?

maehne, 2021-07-13:
Issue has been addressed by PR #354. @klenin: Please reopen if it should still make problems on your side.

closed 2021-07-13 by maehne