logisim-evolution/logisim-evolution #355: Annotation of active low pins

binarycow, 2020-02-25:
Requesting a feature to add annotation of an active low pin.

For example, suppose I have a subcircuit, which has an input which is active low.  On the outer circuit, I simply see the pin name.  On schematic diagrams, it is often drawn with an overbar on the pin name.

In [KiCad](https://kicad-pcb.org/), prefacing a pin name with a tilde (`~`) will render that pin's name with an overbar over it.

Workarounds:

1. On the subcircuit, put an inverter immediately after the input pin...  
    - Adds propegation delay that is not present in the actual circuit
    - Makes it more complicated to implement the circuit and/or transfer it to a different program - I have to remember that the inverter is not actually needed - its just a workaround for active low.
1. Edit the appearance of the subcircuit to add a line over the pin name
    - Manually specified appearance will have to be manually maintained if I later change the circuit.

maehne, 2020-02-27:
In my opinion, this is primarily a question of naming convention. The KiCAD convention, which you suggest, is not suitable, as the resulting port name would not be a valid identifier in VHDL and Verilog. In HDL-based designs, the suffixes `_b` and `_n` are common to denote low-active signals. We use this convention for naming signals in Logisim projects, which are done as part of our digital design courses, even before introducing VHDL itself.

Drawing a bar above the port names in the automatically generated block symbols is certainly doable with reasonable effort, but not high priority. Pull requests are welcome! To make the feature flexible, the list of suffixes leading to drawing the overbar should be configurable via the preferences. It should be also possible to turn this feature off.

binarycow, 2020-03-08:
> Drawing a bar above the port names in the automatically generated block symbols is certainly doable with reasonable effort, but not high priority. Pull requests are welcome! To make the feature [...]

I agree, it should be fairly easy to draw the overbar...  I would totally do a pull request, but my Java skills are basically nil... :(  Maybe eventually I'll figure it out!

closed 2021-07-14 by BFH-ktt1