logisim-evolution/logisim-evolution #356: Identify a "display" for use in sub-circuit appearance

binarycow, 2020-02-25:
Requesting a feature to add a 'display' of some form, that can be used in the sub-circuit appearance, to give an indication of the inner-workings of the sub-circuit from the outer circuit.  This can be useful when a sub-circuit uses a tri-state output - its "value" is not always seen on the output bus, but it can be useful to see what its current value is when debugging.

Multiple memory-type sub-circuits have a 'display' attribute (the `Counter`, `Register`, and similar sub-circuits show their current value).  This feature would simply provide the ability for me to select an 'output' pin as a 'display' pin, and have its current value shown on the sub-circuit, when viewed from the outer circuit.

Workaround:

- Use components with a two-state output.
- Use a controlled buffer to turn the two-state output into tri-state output
- Place a two-state output pin after the two-state output, but before the controlled buffer.  Use this as a 'Debug' display.
- Place a tri-state output pin after the controlled buffer.  Use this as the actual output of the circuit.
- On the outer circuit, connect an 'output' pin to the 'Debug' pin - use this to "peek" at the state of the sub-circuit.

maehne, 2020-02-27:
One simple workaround we use frequently is to create additional output pins for debugging the internal state of a block. We then create a custom symbol, in which we move these debug outputs inside the symbol itself and label them appropriately. When we place instances of these blocks, we directly connect LEDs with these debug outputs. You can even do something similar with bus signals by using, e.g., the binairy to BCD and BCD to seven segment components from the BFH mega functions library to create the individual signals for driving one or more seven segment display(s), which you can directly put on top of the instantiated block if you place the debug outputs in the right pattern. (The BCD to seven segment components already use that approach.)

Implementing such a internal state debug feature in a generic way that avoids debug output ports, is easy to use, and flexible is challenging and will probably require quite some effort. Pull requests are welcome!

binarycow, 2020-03-08:
> We then create a custom symbol, in which we move these debug outputs inside the symbol itself and label them appropriately. When we place instances of these blocks, we directly connect LEDs with [...]

I hadn't thought of that!  I just checked out the BFH seven segment stuff, and that would work great!

BFH-ktt1, 2020-04-02:
In the appearance editor of a circuit there is the possibility to add "dynamic components" (icon completely right in the toolbar). Currently supported are among others LED's, seven segment displays, counters, etc. I know that this feature is not yet completely documented (one can find some of my documentation when looking in the English documentation of the SOC library, but it is far from complete). I merged and extended this feature from @kevinawalsh  that implemented this in his Holycross edition.

closed 2021-07-14 by BFH-ktt1