logisim-evolution/logisim-evolution #357: V3.3: Error: a JNI error has ocurred, please check your installation and try again

joseatienzapiedra, 2020-02-27:
Hello!

I have been using Logisim evolution 2.15 for the last months and everything has been great. I recently tried to run the new 3.3 version but it crashes when I try it.

I use Windows 10 64 bits. I have tried in different computers, with the last version of java and with older versions of java and I have also tried in Windows 7. I Always get the same error:

Error: a JNI error has ocurred, please check your installation and try again

I would really appreciate any kind of help. Thank you sou much!

José Atienza

maehne, 2020-02-27:
What does `java -version` return in the command shell from which you launch the `logisim-evolution-3.3.0-all.jar` file available from the [Releases page](https://github.com/reds-heig/logisim-evolution/releases/tag/v3.3.0) using `java -jar logisim-evolution-3.3.0-all.jar`? On my Windows 10 64-bit machine it returns:
```
openjdk version "13.0.2" 2020-01-14
OpenJDK Runtime Environment AdoptOpenJDK (build 13.0.2+8)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 13.0.2+8, mixed mode, sharing)
```

I suspect that you have several JDK versions installed in parallel on you Windows machine and that a too old version is still in the `PATH`. It is also worthwile to check which JDK / JRE is referenced in the `JAVA_HOME` and `JRE_HOME` environment variables -- if they are defined. Logisim-evolution needs at least a Java 9 JDK. However, I suggest that you install the latest OpenJDK 13 version, e.g., available from [AdoptOpenJDK](https://adoptopenjdk.net/?variant=openjdk13&jvmVariant=hotspot). We use that version for developing and executing Logisim-evolution.

Finally, you may consider building Logisim-evolution yourself using gradle with the help of the instructions found on the [project page](https://github.com/reds-heig/logisim-evolution). Several bugs got fixed since the last release in January. I usually build it using `gradlew shadowDistZip`. That task generates a ZIP archive in the `build/distributions` folder, which contains a handy Batch file wrapper for launching Logisim-evolution more easily, as you can then generate a shortcut on the batch file, which you can place into your start menu or desktop.

maehne, 2020-03-30:
@BFH-ktt1, there has been no reaction by @joseatienzapiedra to my reply. Therefore, I suggest to close this issue assuming that my hints helped him to resolve the JNI error. If not, @joseatienzapiedra can re-open the issue with additional information.

BFH-ktt1, 2020-04-02:
Please reopen when not solved

closed 2020-04-02 by BFH-ktt1

wgaylord, 2020-07-24:
Does Logisim-evelution actually use Java 9 features or is the requirement for Java 9+ just superficial because its what it was compiled with.

maehne, 2020-07-25:
Yes, Logisim-evolution does depend on Java 9+, since we started to remove old, deprecated/unmaintained third-party JAR dependencies, which were originally included in this repository. Now, we try as much as possible to pull them in at build time with gradle. The commit 95a647f1 by @davidhutchens marks the point, where we started to require Java 9.