logisim-evolution/logisim-evolution #363: README: Broken link to a book

klenin, 2020-03-07:
It seems that the book does not exist any more:
http://www.lulu.com/shop/george-self/exploring-digital-logic-with-logisim-ebook/ebook/product-21118223.html

maehne, 2020-03-08:
Thanks for reporting the broken link! This is unfortunately a duplicate of the old open issue #96, we overlooked to resolve, when @grself was so kind to provide an updated version of that old document.

@grself now maintains the documents in two GitHub repositories ([Digital Logic Text](https://github.com/grself/CIS221_Text) and [Logisim-evolution Lab Manual](https://github.com/grself/CIS221_Lab_Manual)). @grself doesn't use the Release functionality of GitHub for the moment, but the compiled PDFs are available in his repositories ([Digital Logic Text](https://github.com/grself/CIS221_Text/raw/master/dl.pdf) and [Logisim-evolution Lab Manual](https://github.com/grself/CIS221_Lab_Manual/raw/master/dl_lab.pdf)). He has also released accompanying videos in a [YouTube channel](http://bit.ly/2KLMcoc).

We have to update our README accordingly.

BFH-ktt1, 2020-03-08:
Should be fixed now, closing issue

closed 2020-03-08 by BFH-ktt1