logisim-evolution/logisim-evolution #375: TCL_Generic

paoloSantinelli, 2020-03-31:
Hi,
I am Paolo, I am a teacher and I use Logisim to teach digital design in my courses.  I am trying to create a TCL_Generic block to access to the serial port of the PC in order the simulated circuit can be able to communicate with external devices.

Please, could somebody give me some help about how to write a simple TCL script to link with the  TCL_Generic block just to do very basic task? I have tried to use the functions "examine" and "force" and the callback refresh, following the Library documentation, but without any success.

I am using Logisim evolution  2.13.8 running on Ubuntu 18.04. and I have installed TCL, ver 8.6.

Thank you

maehne, 2020-04-01:
I am afraid that your request can be only be answered by a member from the REDS institute of HEIG-VD in Yverdon (e.g., @reds-heig, @reds-kgs). To my knowledge, they are the only users of the two components TCL REDS console and TCL generic. I hope that they can provide you with some guidance. We primarily synthesize our design directly on Xilinx or Altera/Intel FPGA boards using the FPGA commander of Logisim-evolution and the vendor toolchain. This gives us much more freedom to link the design with other peripherals.

In any case, I would recommend that you update your logisim-evolution installation to the [latest released version](https://github.com/reds-heig/logisim-evolution/releases), as it contains many bug fixes and new features.

paoloSantinelli, 2020-04-01:
Thank you very much for the tips, I have already downloaded the updated version. Now I try with this version, and, in case, I will ask info at a member from the REDS institute. Usually, We use FPGA boards, but in now each students cant attend the Lab (COVID19) so I am looking for something can be used by the students at their home.

closed 2021-07-14 by BFH-ktt1