logisim-evolution/logisim-evolution #381: Add "Reset window layout" to Window menu to recover from unusable GUI

maehne, 2020-04-06:
The Left panel and therein contained Library and Property panes can be made to disappear by resizing them using the mouse (see screenshots below). Even the main circuit drawing area can be made to disappear (see screenshot below). Unfortunately, it is not possible to make reappear the respective panel/pane, as there's no widget element allowing to resize it back to its initial size. (This is not completely true on all platforms: On Linux and Windows I was able drag back the vertical line to make reappear the left pane or the drawing area. Though, this does not seem to work at all for the design and property panes. The problem seem to be independent of the configured Look & Feel. On macOS, I am not even able to drag the vertical to make reappear the left pane or the drawing area.) 

As the Library and Property panes offer functions, which cannot be reached by other means, Logisim-evolution becomes unusable! Unfortunately, this size change gets instantly saved to the Logisim preferences so that even restarting Logisim-evolution does not help.

The only way to reset the initial window layout is to delete the Logisim settings **when all instances of Logisim-evolution are closed.** Unfortunately, they get stored in different places depending on the OS:

- In Windows the settings are stored in the registry under the key "Computer\HKEY_CURRENT_USER\Software\JavaSoft\Prefs\com\cburch".

- In macOS the settings are stored in the file "~/Library/Preferences/com.cburch.logisim.plist".

- In Linux, the settings are saved in the folder "~/.java/.userPrefs/com/cburch/".

I did not find any other way to do this through the Logisim-evolution GUI even though, I checked the program preferences, project options, Edit menu, and Window menu for appropriate options. Even the promising button "Project -> Options -> Reset -> Reset All Settings" won't help in this matter. Anyway, I would suggest adding a more easily to find way to restore the initial window layout, e.g., by adding a menu item "Reset window layout" to the Window menu.

[image]

[image]

[image]

[image]

maehne, 2020-04-06:
Maybe, making these different panes to disappear could be suppressed alltogether by forcing a minimum size to each of them?

BFH-ktt1, 2020-04-06:
The dissapear is a "wanted feature" that is used in Logisim. Hence preventing the minimal/maximal size will go against some of the features used. I added a button in the preferences -> window with which the default layout can be restored.

commit 22c506e6 (2020-04-06): Merge pull request #383 from BFH-ktt1/bugfixes

Fix #381
closed 2020-04-06 by BFH-ktt1