logisim-evolution/logisim-evolution #382: Improve German translation of FPGA Commander messages

maehne, 2020-04-06:
Thanks for internationalising the FPGA Commander GUI! The German translation requires work, as some automatically translated messages have turned out to be either non-sensically or funny. Examples from `src/main/resources/resources/logisim/de/fpga.properties`:

- [ ] "Sie haben ein Blatt in Ihrem Design mit einem leeren Namen gefunden. Dies ist nicht erlaubt, bitte geben Sie einen Namen an!"
- [ ] "Ich habe eine bewachte Uhr gefunden. Vorsicht, die echte Hardware funktioniert möglicherweise nicht richtig!"
- [ ] "Einen Verteiler gefunden, der keine Signale vom Bus zum Lüfter transportiert."
- [ ] "Ich habe eine Spüle ohne Quelle gefunden!"
- [ ] "Blinkende Bitdatei"
- [ ] "Während des Blinkbetriebs ist ein Fehler aufgetreten."
- [ ] "Fehler beim Blinken der Bitdatei."
- [ ] "Ich habe %d angebrachte Boards gefunden. Wählen Sie aus, welche programmiert werden soll....."
- [ ] "Platzierung und Fräsung Design"

BFH-ktt1, 2020-04-06:
This is true for most languages except for the English version. From now on I will refrain from using Deepl for translation as it is probably less annoying to have some English text as some funny no sense making "own" language text.

maehne, 2020-04-06:
Well, there is still some value in having the DeepL translations as a starting point. We native speakers just need to find and take the time to review those translations and improve them. That's why I created this issue.

We may discuss whether it is better to simply put the English phrases as placeholders into the translation files for the other languages. For users this might be more convenient, but less motivating for potential contributors of good translations.

maehne, 2020-04-06:
- [ ] "Altera-Synthese, P"

BFH-ktt1, 2020-04-06:
When no translations are available yet, we should not specify any thing in the for example German files, as logisim will take in this case automatically the english version. This is the reason why I mark them by:
#==> <key>
Such that somebody working on them knows that there is not yet a translation available, and that they can find a description of the <key> in the English version to be translated

R3dst0ne, 2021-06-03:
> Well, there is still some value in having the DeepL translations as a starting point. We native speakers just need to find and take the time to review those translations and improve them. That's why I created this issue.
>
After working on the translations for my [Pull request](https://github.com/logisim-evolution/logisim-evolution/pull/608), I can confirm this statement. It is way easier and faster to search for errors than to translate everything by your own.

R3dst0ne, 2021-07-01:
I fixed/added a lot of translations in bd20be0786e72d29c481e5cb438f73aae70b67fe.
Did I miss something?
If not this issue can be closed.

maehne, 2021-07-01:
Thanks for your improvements to the German localisation @R3dst0ne!

closed 2021-07-01 by maehne