logisim-evolution/logisim-evolution #384: Missing components

hugolundin, 2020-04-09:
I was looking for the "Power On Reset" component that *should* be in "Wiring". It's simply not there for me. 

I am using the application on macOS 10.15.4. I built it myself by cloning the master branch and running `./gradleW createApp`. When I look in `/logisim-evolution/src/main/java/com/cburch/logisim/std/wiring` of the repository I built there is a file called `PowerOnReset.java`. 

Any suggestions?

maehne, 2020-04-09:
Well, the "Power On Reset" component is simply called "POR" and just located below the "Clock" source component in the "Wiring" library of Logisim-evolution. I checked versions 3.3.0, 3.3.1, and 3.3.3. Did you maybe just overlook it?

hugolundin, 2020-04-09:
Yep, I did 😄 Thanks!

closed 2020-04-09 by hugolundin