logisim-evolution/logisim-evolution #385: Wire value X labeled as Z for hi-Z

bmyerz, 2020-04-09:
I've always had to explain to my students that the X you get from poking a blue wire is not related to "don't cares" in truth tables. Rather the X's are indicating hi-Z state. I'm interested in whether there would be support for showing the X's as Z's instead to reinforce this?

BFH-ktt1, 2020-04-09:
Dear @bmyerz ,
I completely agree with the fact that a three-stated wire should not be called "X", as "X" can have two different meanings:
1) being don't care.
2) Being a short circuit (as in the IEEE1069 specifications)
I will look into the issue and try to solve this issue, thank you for your Issue request.

BFH-ktt1, 2020-04-13:
Updated in my latest push, documentation needs still to be adapted, hope this solves the problem