logisim-evolution/logisim-evolution #393: Consider integrating simplified Chinese locales recently merged into Logisim ITA

maehne, 2020-04-19:
Simplified Chinese locales were added in [PR #14](https://github.com/LogisimIt/Logisim/pull/14) to [Logisim ITA](https://github.com/LogisimIt/Logisim).

ManiaciaChao, 2020-04-21:
Well, considering the difference in the text between the two versions, I need to spend some time rechecking and updating the text to make the translation more suitable for `logisim-evolution`. 
I will make a PR as soon as possible.

maehne, 2020-05-06:
@ManiaciaChao: Thanks for offering your help in this area. Logisim-evolution incorporates almost all improvements done by the [Logisim ITA](https://github.com/LogisimIt/Logisim) developers as well as many from the [Logisim-evolution (Holy Cross)](https://github.com/kevinawalsh/logisim-evolution) edition by Kevin A. Walsh. In addition, it contains several additional improvements especially related to downloading the design on FPGA boards and SoC design. It is a bit unfortunate that Logisim development is carried on in separate forks since Carl Burch stopped developing the original version. Nevertheless, we are very open for collaboration and external contributions.

R3dst0ne, 2021-11-07:
@maehne @ManiaciaChao @legion-null 
Since #1282 got merged, is this issue still relevant?

closed 2021-11-07 by BFH-ktt1