logisim-evolution/logisim-evolution #396: Chip Select with RAM

rhodeon, 2020-05-03:
The original Logisim has a chip select port which enabled or disabled the entire component, and it was level triggered. 
There doesn't seem to be an equivalent port in Evolution. Is there a way to achieve this?

maehne, 2020-05-06:
Well, you have the dedicated `Write enable` and `Output enable` inputs. How about connecting your `Chip select` signal directly to `Output enable` and using an AND gate in front of the `Write enable` to generate from the `Chip select` and `Write` control signals the `Write enable` signal?

You are aware that you can switch the appearance of components in their property pane to get back the classic Logisim look?

rhodeon, 2020-05-08:
> Well, you have the dedicated `Write enable` and `Output enable` inputs. How about connecting your `Chip select` signal directly to `Output enable` and using an AND gate in front of the `Write [...]

Yeah. That was coincidentally how I ended up mimicking the pin.
As for the classic look, switching to it just changed the interface of the RAM and didn't create a chip enable pin. I suspected this would be the case, though.

MarcinOrlowski, 2021-07-13:
Can we please try to evaluate if we are going to add this or not?

closed 2021-07-14 by BFH-ktt1