logisim-evolution/logisim-evolution #397: Circuit Appearance isn't persistent across files.

rhodeon, 2020-05-05:
Creating a circuit and editing it's appearance works when using it in other circuits in the same file, but not when loaded as a library in another file.

For example, I made an implementation of the 7489 RAM IC and edited its appearance to be:
[image]

It appears this way when used in other circuits in the same file.

But when loaded as a library in another file, it reverts to its appearance before modification:
[image]

I tried it out on the original Logisim and it doesn't exhibit this behaviour. Version 2.14.6 of Evolution doesn't have this problem either.

maehne, 2020-05-06:
Thanks for reporting the bug! I can reproduce the issue (see attached test case).

@BFH-ktt1: Could you have a look?

[bug397_testcase.zip](https://github.com/reds-heig/logisim-evolution/files/4585186/bug397_testcase.zip)

BFH-ktt1, 2020-05-06:
This is my issue #334 , and on my todo list

rhodeon, 2020-05-08:
Thanks for the response. Do you currently have an idea of the ETA for the next update?

Triavanicus, 2020-07-15:
Where it changes
com.cburch.logisim.gui.menu.ProjectLibraryActions:177
package com.cburch.logisim.proj.Project:210
package com.cburch.logisim.file.LogisimFileActions.doIt (412)
com.cburch.logisim.fileLogisimFileActions.repair:435

It seams that when you load a library it will try to repair it, overwriting the appearance with the default appearance.

Triavanicus, 2020-07-15:
If you create a method Circuit.java called setAppearance and set the appearance of the NewCirc, it won't work, at all, if circumvent the whole process and on line 460 of LogisimFileActions.java change it to `added.add(circ)` it will work just fine. So maybe if there was a way to repair it only if necessary, or something it will fix this bug.

BFH-ktt1, 2020-07-15:
@Triavanicus 
Thanks a lot for your investigation, I took a quick look and remembered why the repair function is in place:
When loading a library also the standard libraries are loaded with the .circ file. If we would just do added.add(circ) all the components inside the circuit (except for the wires) would be linked against the libraries in the loaded "logisimfile" instead of the ones available in the "project". Meaning that we would have for example 3 different objects for a LED. In HDL-generation this causes severe problems.
The "repair" function will create a new circuit that links against the "project" instead if the "logisimFile". I forgot to update the appearance, and have to look into the details "how to copy" the correct appearance in the repair function (this might or not be a bit more involving).

BFH-ktt1, 2020-07-15:
@Triavanicus , @rhodeon , I hope that this fixes the problem, I could not test all situations.
What I did is instead of making a new circuit, I replaced all factories of the components. This should be proper.

Triavanicus, 2020-07-15:
It seems to work in my case.

Triavanicus, 2020-07-20:
Just tested the old 7400 library, and it uses the correct appearance. This issue should be closed.

rhodeon, 2020-07-23:
Can confirm it works as intended. Will be closing the issue now

commit 3a259fa3 (2020-07-15): Merge pull request #421 from BFH-ktt1/chronogram

This should fix #397, hopefully I did not introduce another bug
commit 6ac037ae (2020-07-15): This should fix #397, hopefully I did not introduce another bug
closed 2020-07-23 by rhodeon