logisim-evolution/logisim-evolution #398: Start Chronogram button missing.

rhodeon, 2020-05-08:
The current chronogram window in version 3.3.4 appears thus:
[image]

Here's a screenshot of the window in version 2.14.6:
[image]

It's currently missing the bottom bar with the "Start Chronogram" button. As it stands now, there's no way to use the chronogram.

tombjerck, 2020-06-12:
This error was introduced by commit 1aa577d3b42b210fd734ce4fc356654d72d06b6f from Mar 23 2019. This is while running openjdk version "11.0.7" 2020-04-14.

maehne, 2020-06-12:
Thanks @tombjerck for identifying the specific commit, which introduced this commit! @BFH-ktt1 made these modifications and I know that he is not yet finished with porting back the improvements done by @kevinawalsh in his Logisim-evolution Holy Cross edition. It may take several weeks before he may find time again to look into this issue, as the spring term exam period lies ahead with a high work load after a particularly charged semester.

However, if you have a solution to propose to fix this or other issues, don't hesitate making a pull request against the develop branch!

BFH-ktt1, 2020-06-16:
Yes, the chronogram is broken for the moment. Currently I'm finishing up some big rewrites of the FPGA-commander. After this I will attack the chronogram and merge the nifty features Kevin Walsh added to his edition of logisim. Please be a bit patient.

Bravo555, 2021-01-10:
Hello, is there any progress on this issue?

maehne, 2021-01-10:
Unfortunately, not yet due to our high work load over the past months. I know that @BFH-ktt1 will be quite busy the coming months with teaching duties. So it is hard to give an estimate when he will get around to fix this issue. Pull requests are welcome!

lightonflux, 2021-01-29:
Is this "just" an UI issue?

rikard1234, 2021-03-30:
Hello, any progress on issue?

BFH-ktt1, 2021-03-30:
As I see currently my schedule there will be no activity on this part of the code before July 2021. The pandemic situation has us all given lots more of work as under normal circumstances.

rikard1234, 2021-03-30:
Alright thanks for response, may I ask if you could tell what was the last version in which that button worked, or maybe share that version? Pretty desperate to get it because of college project

Bravo555, 2021-03-31:
@rikard1234 it was working fine for me on version 2.14.6

BFH-ktt1, 2021-04-06:
You should now have the great timing diagram that @kevinawalsh made in his holycross version. There might still be bugs, so please use this thread to report them. I will publish one of these days a version v3.4.4 including these enhancements.

maehne, 2021-07-12:
@rhodeon, @Bravo555, and @tombjerck: Did you have a chance to test the new timing diagram functionality? Can this issue be closed as fixed?

Bravo555, 2021-07-13:
Yes, tested version 3.5.0 and everything works on my end, the new timing diagram looks really good! As for me, the issue can be closed. Thank you for the fix!

maehne, 2021-07-13:
Thanks for the feedback @Bravo555!

PR #541 (2021-04-06): Timing diagram (chronogram)
After a long time of merging I finished the great improvements that @kevinawalsh made (amongst others) to the chronogram and simulator core. Most of these pushes are a direct copy of @kevinawalsh 's work just adapted to the current implementation of logisim. Please note that:
1) There might still be bugs.
2) I did not completely format correctly all source files, as I wanted to give you the functionality.
3) All keys added and removed are only performed in the English files.
Some help to clean up is very welcomed.
This fixes #398.
closed 2021-07-13 by maehne