logisim-evolution/logisim-evolution #401: Write enable pin missing from flip-flops

benburk, 2020-06-01:
The write enable pin is missing from the different flip flop circuits. These were present on the original circuits as shown [here](http://www.cburch.com/logisim/docs/2.3.0/libs/mem/flipflops.html).

ReGuess, 2020-06-13:
See #128 
Although I will say that either it ought to be brought back, or the documentation ought to be updated regarding this, and ought to explain how to recreate all four kinds of flip-flops using registers. The only one that's immediately obvious is the D flip-flop; and even still, registers lack the "preset/pin to 1 asynchronously" input pin available on the flip-flops.

BFH-ktt1, 2020-06-16:
The documentation on the flipflops is not up to date, still on the todo list.  I agree that on the registers the asynchronous preset pin is missing and opt for this option rather than adding them to the flipflops. But I'm open for suggestions. Adding and enable pin to the flipflops in a sense is dubious as it can be implemented by a gated clock (in general not a good option) or by a feed-back that keeps the flipflop in it's current state (prefferred method). Please let me know what you think, it can be re-introduced by using an attribute for example.

Triavanicus, 2020-07-15:
Here is an example on how to do this.
[FlipFlops.zip](https://github.com/reds-heig/logisim-evolution/files/4925021/FlipFlops.zip)

LordDecapo, 2020-08-02:
[image]
The FlipFlops.zip above has a gated clock. If you want to put your project onto an FPGA via the internal FPGA tools, this will throw an error "Gated Clock" and won't let the compile finished. The image I have included has a non-gated clock version of what Ben is looking for [hi btw] and will take up less resources on an FPGA.

closed 2021-07-14 by BFH-ktt1