logisim-evolution/logisim-evolution #406: Programming hardware cable not detected

carolineeeeeee, 2020-06-24:
Hi,

I have been trying to download the circuit design in Logisim onto the Altera DE1_SoC board that I have. I am using the file containing the information of the board and the pins that I created and testing it with a simple AND-gate like this:
[image]

Everything works fine and it compiles successfully but at the last step when I click on *Yes, download*, I get an error in the console: `Error: Programming hardware cable not detected` and then in Errors:`Failed to Download design, did you connect the board?`.

I can confirm that the board is connected fine because I am able to directly program it using Quartus but for some reason, Logisim is unable to detect it. I am not quite sure which part went wrong and how to fix it. 

Here's the information I used for the DE1_SoC board:
```xml
<BoardInformation>
      <!--This section decribes the FPGA and its clock-->
      <ClockInformation FPGApin="PIN_AF14" Frequency="50000000" IOStandard="LVTTL" PullBehavior="Float"/>
      <FPGAInformation Family="Cyclone V" FlashName="" FlashPos="2" JTAGPos="1" Package="FBGA 896" Part="5CSEMA5F31C6" Speedgrade="6" USBTMC="false" Vendor="ALTERA"/>
      <UnusedPins PullBehavior="Float"/>
</BoardInformation>
```
Thanks.

BFH-ktt1, 2020-06-24:
Dear @carolineeeeeee,
As the board you are using has a SOC FPGA (also includes the ARM-Cores) it can be that the FPGA cannot be configured without the ARM's being set-up. I know that this is the case for the Xilinx-Zynq platform. It is on my TODO-list to check this out. As I also have the same board (DE1_SoC) in my office I am happy to see where the problem comes from. If you could post the complete board description file you made for the board I am happy to try it out to see where the problem lies, and how to fix it. I guess that the call to the programmer software of quartus is missing some command line options that are required for the ARM-cores.

carolineeeeeee, 2020-06-24:
Hi @BFH-ktt1,
Thanks for your reply! I have been trying to figure this out for a while now... It would be so helpful if you could try it out. 
Here is the board description file I made (I changed it to txt so that it can be attached here):
[DE1_SOC.txt](https://github.com/reds-heig/logisim-evolution/files/4828325/DE1_SOC.txt)
Thank you!

BFH-ktt1, 2020-06-25:
Dear @carolineeeeeee ,
I found the problems, one was a bug in logisim, that has been fixed in my latest push, and the other was a problem in the board-description file as the FPGA is the second device in the JTAG-chain (hence JTAGPos="2" instead of "1").
Furthermore I detected that you swapped the locations of all the elements, hence if you map in the FPGA-commander on the left-most 7-segment display, the real hardware will have the rightmost 7-segment display lighting up) I fixed this only for the seven segment display in the attached file, please feel free to fix the others. (note you can now drag the components around in the board editor).
I close the issue for now, please feel free to reopen it when the problem is not resolved.
[DE1_SOC.txt](https://github.com/reds-heig/logisim-evolution/files/4832145/DE1_SOC.txt)

commit 017abc31 (2020-06-25): Merge pull request #408 from BFH-ktt1/chronogram

Fixed #406 and some other bug fixes
closed 2020-06-25 by BFH-ktt1