logisim-evolution/logisim-evolution #410: RAM/ROM does not load content from file properly

fuzun, 2020-06-27:
When you set address bit width to 8, and data bit width to 48 and try to load this file, it does not load properly:
It probably happens when data bit width is greater than 32 bit
```
v3.0 hex words addressed
00: 00a500000000 00000c000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
08: 000180404084 00000c000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
10: cde9a30f8280 00000c000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
18: 00001420008c 87000c000000 000000000001 000000000000 000000000000 000000000000 000000000000 000000000000
20: 218000000000 00a50c240000 000000000001 000000000000 000000000000 000000000000 000000000000 000000000000
28: cde9bf0f80a0 00000c000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
30: cde9bf0f80b0 00000c000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
38: 5b7f00000000 cde9ac0f8280 cde9a30f8282 000000000001 000000000000 000000000000 000000000000 000000000000
40: 5a7e00000000 cde9ac0f8280 cde9a30f8282 000000000001 000000000000 000000000000 000000000000 000000000000
48: cde9bf0f80b8 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
50: cde9bf0f80c0 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
58: cde9bf0f8090 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
60: cde9bf0f80d0 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
68: cde9bf0f80d8 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
70: 250000010000 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
78: 310000010000 00000c00000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
80: 310000010000 00000000000b 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
88: 870000200084 250000010000 000000000001 000000000000 000000000000 000000000000 000000000000 000000000000
90: 860000000000 250000220000 000000000001 000000000000 000000000000 000000000000 000000000000 000000000000
98: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
a0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
a8: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
b0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
b8: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
c0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
c8: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
d0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
d8: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
e0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
e8: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
f0: 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
f8: 00a500140000 000000000001 000000000000 000000000000 000000000000 000000000000 000000000000 000000000000
```

https://i.imgur.com/tZ21amK.png

fuzun, 2020-06-27:
@BFH-ktt1 I'd appreciate if you take a look at this. Resolving this problem is very crucial for our task. A great effort was made to create a circuit that depends on a ROM that is capable of loading 48-bit width content. We have limited time to derive a solution.

fuzun, 2020-06-27:
Probably related: https://i.imgur.com/Z9PXqCX.png
In hex editor, I can not edit 16 bits from the MSB when the bit width is high. They always stay 0xf.

BFH-ktt1, 2020-06-28:
This was resolved by #394 in the development branch. Please use the version in the development branch.
In the future please specify the version of logisim you relate your issue to and always try the version in the development branch, because maybe the issue is already resolved.
Closing the issue for now, please reopen if you feel the issue is not correctly resolved.

closed 2020-06-28 by BFH-ktt1