logisim-evolution/logisim-evolution #414: Simple Verilog and VHDL exports.

PeterHindes, 2020-07-07:
I just want to be able to export my design as a verilog or vhdl file. I don't want to have to specify a board and mark which parts of the board I am using. I want to give an input a name and have it listed as that in a verilog output.

Also somewhat related, could someone give me a brief description of how to add a part to the list of compatible parts. I want to use the controlled buffer in my fpga designs and I think it should be simple for me to implement.

maehne, 2020-07-07:
> I just want to be able to export my design as a verilog or vhdl file. I don't want to have to specify a board and mark which parts of the board I am using. I want to give an input a name and have it listed as that in a verilog output.

@BFH-ktt1 has recently reworked the whole FPGA commander part of Logisim-evolution. He should have a good idea how much effort the implementation of your proposed enhancement would cause. However, exposing people too much to this generated VHDL code might not be the best idea, as it is basically a gate-level structural description, which is much more complex than an equivalent description at RTL.

> Also somewhat related, could someone give me a brief description of how to add a part to the list of compatible parts. I want to use the controlled buffer in my fpga designs and I think it should be simple for me to implement.

Check out the implementation of the different components in `logisim-evolution/src/main/java/com/cburch/logisim/std/`. The HDL generation for each component is encapsulated in a `<Component>HDLGeneratorFactory` class, which is derived from `AbstractHDLGeneratorFactory`. That factory is returned by the HDLSupportedComponent() member function of the corresponding component class.

However, Logisim-evolution cannot handle inout ports, which would be needed to benefit from controlled buffers in a synthesisable design, see issue #253. I suggest to have a look at the recently enhanced PORTIO component, which is now supported for synthesis. You'll need to build from the [develop](https://github.com/reds-heig/logisim-evolution/tree/develop) branch to do so.

BFH-ktt1, 2020-07-08:
Unfortunately a part of the HDL-generator depends on the board information (especially the clock frequency) to be able to determine the values for the clock divider(s) that generate the tick. Note that the HDL generated is not "just" a description of your logisim circuit, some components are modeled multi-cycle (like for example the RNG and others). Furthermore the HDL-generator of logisim "translates" the circuit such that it works exactly as in logisim (same frequency, etc.). It would be quite some effort to build in a "translator" that generates the HDL 1:1. This is also not the goal of the HDL-generator in logisim. If you want to have just a schematic entry for your FPGA, both Quartus and Vivado provide you with this possibility.

I implemented the HDL-generator for the controlled buffer, but please be carefull with them, it would not be the first time that I see holy smoke coming out of an FPGA due to wrongly used tri-state buses inside the FPGA, or a none-functioning system as the synthesys tool was not able to correctly transfer the tri-states to non-tri-state.

PeterHindes, 2020-07-13:
Yes, I thing a good way to do this might be to have all the components be sub modules, and the wiring in a simple hdl generation describing the wires that the user can understand. The clock divider should be as simple as asking the user what pin the clock is on and what speed it is and then put the correct values in to a clock divider.

BFH-ktt1, 2020-07-13:
Please feel free to modify logisim for this purposes and ask for a push-request in the development tree.

closed 2021-07-14 by BFH-ktt1