logisim-evolution/logisim-evolution #417: Estimated release date for a new version?

carolineeeeeee, 2020-07-09:
Hi,

Currently, at our University, we are teaching an introductory hardware course that teaches students the basics of building circuits. We are hoping to use Logisim as the tool to help students learn about the circuits visually. 

The reason we are a bit concerned about the release date is that we only have the DE1 SoC boards in our labs. And from #406, the downloading design on the DE1 board is just added to the development branch (Thank you so much for that). Since we are using this tool for a course, we are hoping to have a stable version of the tool for the students to use. Therefore we are wondering if you would be willing to release a newer version of Logisim containing the update in #406 at some point in the near future. 

One more small question: currently Logisim supports testing for combinational logic using the test vector files, I am wondering if a similar testing mechanism exists for sequential logic? I know that we can directly simulate on the canvas but is there a way to systematically test a sequential circuit in Logisim? 

Thank you so much!

BFH-ktt1, 2020-07-09:
Done, the pre-release 3.3.6 is available (latest head of the develop branch).

About your question: As I did not look into/have used the test-vector part of the code I cannot answer correctly to your question. 
There was the chronogram, but this is still to be updated/merged from the holycross version of @kevinawalsh, which is on the top of my todo list of things to do. But 1) as I am uptill the 19th of July in the exam period of my university, and 2) Covid pushed stuff into the August, period I have limited bandwidth for the moment.

BFH-ktt1, 2020-07-13:
I close the issue, hope it helped you out. please reopen if you feel the issue is not fixed.

closed 2020-07-13 by BFH-ktt1