logisim-evolution/logisim-evolution #426: Triple XOR Input

Isam-Ilyas29, 2020-07-29:
I am using the latest release (3.3.6) and am having an issue where my triple input XOR gate, when receiving 1 through each input, where the data bit is a 0 when it should instead be a 1.

Schematic:
[image]

Properties:
[image]

LordDecapo, 2020-07-31:
You have the Multiple Input Behavior set to "when one input is on" when for the behavior you are wanting, you need to have it set to "when an odd number are on".
That will give you what you want.

maehne, 2020-07-31:
@BFH-ktt1: This issue can be closed as resolved.

closed 2020-07-31 by BFH-ktt1