logisim-evolution/logisim-evolution #428: Combinatorical Analysis generates non-minimized circuits, and refuses to connect multiple inputs to a output.

jay20162016, 2020-08-06:
For example, this is part of the circuit generated:
[image]
This circuit could be minimized by having the final NAND connect to the output of the previous circuit twice. 
Here is a truth table to reproduce the output:
```
# Truth table
# Generated from circuit test

inpu[2..0] | outpu[9..0]
~~~~~~~~~~~~~~~~~~~~~~~~
   000     | 0001110011 
   001     | 0100111111 
   010     | 1010000000 
   011     | 0111110011 
   100     | 0000001000 
   101     | 1000010110 
   110     | 0011100001 
   111     | 1100001100 
```

BFH-ktt1, 2020-08-07:
Dear @jay20162016, I am unable to reproduce this effect in logisim-evolution V3.3.6. From what I can see from the circuit you are using an old version of logisim-evolution (probably a 2.x) .
Could you:
1) Please try your circuit again in the latest version of logisim-evolution.
2) Please indicate for which bit of the outpu this circuit is generated
With this info I can than trace down the problem. I close the issue for now, please reopen when the problem still persists in the latest version of logisim evolution.

closed 2020-08-07 by BFH-ktt1

jay20162016, 2020-08-07:
I am using the new version, but I cannot reproduce it.