logisim-evolution/logisim-evolution #431: N-channel transistors does not work - do not pass a value through the transistor in version logisim-evolution 3.3.6

hermanborstlap, 2020-08-13:


maehne, 2020-08-13:
Please provide a sample circuit demonstrating your issue and describe your expected behaviour. The MOSFET models implement a very simplified behaviour, which requires great care when placing them in a circuit so that they are properly oriented, because values are only propagated into one direction. The small green arrow helps with that.

hermanborstlap, 2020-08-13:
Thank you for responding. Attached my exported cct. As a test I simply connected an N-channel transistor with its drain to power constant 1 and the source to a pull down resistor to ground. The gate is connected to an input pin (ctrl-5) and when it is made high the transistor does not switch on. I am familiar with the arrow showing the direction of propagation for a value to pass through. In logisim-evolution version 2.15 I designed my own 2 and 4 inputs AND, OR, NAND, NOR, XOR, XNOR circuits and they work perfectly fine. I copied (by connecting the same components and connections) of the the same 2-input and 4-input AND, OR etc. circuits in version 3.3.6 and cannot get the transistors to switch on and pass a value through. I have subsequently noticed that the P-channel transistor does the same except intermittently they DO switch a value through, but not the N-Channel transistors. 
[image]

BFH-ktt1, 2020-08-14:
The problem with Mosfets is that they are analog circuits and behave differently given a common-drain, common-source, or common-gate connection. As logisim is a pure digital simulator these behaviors are not implemented.
To come back to your circuit,tere is one problem, and there is one solution to resolve your issue:
First the problem: You made an drain follower instead of a digital gate as you put the p-channel Mosfets in the pull-part and n-channel in the push-part (digital gates always have p-channels in the push and n-channel in the pull to have the fastest switching time and to force that the output is always 1 or 0). To implement a and-gate you have to use 6 transistors. (4 to implement an NAND and 2 for the inverter to make an AND)
Then to the solution of your behavior: As you can see your "push" part can transport correctly the 1 to the output "AND_2b" (see direction of the arrow). However, your "pull" circuit cannot transport the 0 to the output "AND_2b" as the arrow goes from AND_2B to GND. Flipping these two transistors such that the arrow is in direction GND > AND_2b will resolve the problem.

closed 2020-09-04 by BFH-ktt1