logisim-evolution/logisim-evolution #433: How to add a translate?

GGmyfriendXD, 2020-08-14:


BFH-ktt1, 2020-08-14:
Dear @GGmyfriendXD ,
In the path src/main/resources/resources/logisim/ you will find the language directories (for example en for english, es for spanish, etc.). Inside these directories you will find the *.properties files that contain the language specific translations. Each of these files contain:
key = description
Where the key requires to be kept as is(as this is used in the sources to fetch the language specific description) and the description in the specific language.
if you find an entry like:
# ==> key =
This means that a language specific description is not yet available and that logisim takes the English (default) description for this key which you can find in the same file in the "en" directory.
To add a new language, this is a bit more involved. Just let me know and I prepare the "*.properties" file for you.
Note that the property files are in UTF-8
Hope this answers your question.

MarcinOrlowski, 2021-07-08:
@maehne could that be moved to Discussions as well?

maehne, 2021-07-08:
@MarcinOrlowski: Sure, that makes sense.

closed 2021-07-08 by maehne