logisim-evolution/logisim-evolution #434: N-Channel Transistors does not pass a value through in version 3.3.6

hermanborstlap, 2020-08-14:
Thank you Maehne for responding at first. Attached my exported cct. As a test I simply connected an N-channel transistor with its drain to power constant 1 and the source to a pull down resistor to ground. The gate is connected to an input pin (ctrl-5) and when it is made high the transistor does not switch on. I am familiar with the arrow showing the direction of propagation for a value to pass through. In logisim-evolution version 2.15 I designed my own 2 and 4 inputs AND, OR, NAND, NOR, XOR, XNOR circuits and they work perfectly fine. I copied (by connecting the same components and connections) of the the same 2-input and 4-input AND, OR etc. circuits in version 3.3.6 and cannot get the transistors to switch on and pass a value through. I have subsequently noticed that the P-channel transistor does the same except intermittently they DO switch a value through, but not the N-Channel transistors.
AND_gate_2IP

hermanborstlap, 2020-08-14:
[image]

maehne, 2020-08-14:
This is a duplicate of issue #431. 

@hermanborstlap: Please, continue the discussion in the original issue. @BFH-ktt1 provided you there already with feedback explaining the behaviour observed by you and the necessary modification.

@BFH-ktt1: You can close this issue.

closed 2020-08-14 by BFH-ktt1