logisim-evolution/logisim-evolution #437: False failures

0x7C8, 2020-09-03:
I have made a circuit in Logisim and have opened it in Logisim Evolution for a verification.
It seems that testing vector provides false failures when testing a bunch of cases. But when I have tested failed cases separately, they have passed.

Here is a circuit with a test vector files. In 2_3.txt 2 cases fail, meanwhile when testing them separetely (2_3_fails.txt) they pass.

[circuit.zip](https://github.com/reds-heig/logisim-evolution/files/5165663/circuit.zip)

BFH-ktt1, 2020-09-04:
dear @0x7C8 , I confirm your findings and thank you for the detailed test case. 
I'll put it on my todo list, but with the current covid situation I cannot guarantee a quick response time

0x7C8, 2020-09-08:
Upon further analysis, I found out that I made a mistake in my test vector which caused failures. Sorry about that.

0x7C8, 2020-09-08:
What caused my confusion is that the test output overwrites expected output value (which is colored red in Test Vector window).

closed 2020-09-08 by 0x7C8