logisim-evolution/logisim-evolution #439: Logisim Holycross (HC) fork looks better in some aspects

John-Lluch, 2020-09-05:
Hi, Congrats for this excellent work on the original Logigim!.  I am using it for homebrew CPU simulation and learning purposes. I also looked at the Logisim Holycross fork [https://github.com/kevinawalsh/logisim-evolution](https://github.com/kevinawalsh/logisim-evolution) and found it has some important improvements

In particular, I found that simulation with the Logisim HC fork is a better experience in many aspects. There's a list of the most significant improvements for me:

- Much faster. Synchronous, clock based simulations can be run at much faster frequencies. Complex long simulations end in a much shorter time, allowing for more productivity. It almost feels a different simulation engine

- Step Simulation shows the changed signals and inputs on top of the screen, helping you to make sure all signals have propagated before you make an input changes for further testing.

- Project can be seen hierarchically in Simulate mode, which helps navigating complex projects to find issues and hierarchy deep simulation values

- Circuit names and labels in Inputs/Outputs/Tunnels still allow any characters like the original Logisim. Among others, this allows the usual naming convention of some old circuitry, for example "/EN" for "Low Enable"   

- Mac app bundle directly available for download.

Since your fork is the most active one, and after all the HC fork was derived from yours, I would quite like to see such improvements added here, so I wonder if you have plans to incorporate the above enhancements to your work.

Thanks!

BFH-ktt1, 2020-09-06:
Dear @John-Lluch,
Thank you very much for your nice words! In fact there is an effort going on to merge the great improvements @kevinawalsh has made to his Holycross version back into Evolution. This has a bit come to a hold due to the Covid situation, but I am planning in the near future to continue this merging (in my limited time available).

maehne, 2020-09-07:
@John-Lluch: As @BFH-ktt1 pointed out, merging back improvements from the Holycross fork is in progress, but hampered by our limited time. Pull requests are therefore always welcome! Regarding your last three points:

> * Project can be seen hierarchically in Simulate mode, which helps navigating complex projects to find issues and hierarchy deep simulation values, in a rather convenient way.

I think this feature is already in place at least in the develop branch. You can switch between design library view and simulate view using the tabs in the upper left pane.

> * Circuit names and labels in Inputs/Outputs/Tunnels still allow any characters like in the original Logisim. Among others, this allows the usual naming convention of some old circuitry, for example it would allow an input to be labeled "/EN" for "Low Enable" .

Circuit names and port labels are constrained on purpose to ensure valid identifiers in the generated VHDL/Verilog net lists for downloading your designs on FPGA boards. Therefore, we also encourage our students from the beginning to use common VHDL/Verilog naming conventions for circuit instances and ports. If you would like to implement an option to turn this off and instead rename identifiers were needed, you are welcome to make such a pull request.

> * Mac app bundle is directly available for download.

You can already build a Mac app bundle and DMG using `gradle`. Providing them alongside the JAR archive as direct download is a good suggestion. In addition, RPM/DEB packages should be also provided for Linux and an MSI installer for Windows. This functionality was unfortunately lost when Oracle removed `jarpackager` from JDK 9 without providing a working replacement. JDK 14 now finally provides `jpackage` as an incubator module, but we did not yet find the time to test it and make the necessary changes in our gradle build system. Ideally, building the release files should be automated using a CI/CD pipeline. However, we did not yet get around to do this. Contributions in this area are very welcome!

John-Lluch, 2020-09-08:
Hi @BFH-ktt1 and @maehne

Thanks for your replies. 

It's good to know that the said improvements are being looked at for incorporation in this fork. I made contributions in the past to big projects such as the LLVM compiler, and I wish I could contribute with pull requests here, but unfortunately I'm not versed in java or java tools, so I can't really offer my help here. 

For testing this, I just downloaded the "compiled stable" .jar file that is linked in the readme file, and ran it as is on a Mac. The 'About..' menu states "Logisim-evolution Version 2.14.6, (C) 2014", but I now realised that there's a more recent version in the repo, so I might be missing some of the latest improvements. In any case, it would be great for users who are not particularly aware of java tools or toolkits, that you update that link if there's a more recent version considered 'stable'.

maehne, 2020-09-13:
Version 2.14.6 is very old. Did you check out the [releases](https://github.com/reds-heig/logisim-evolution/releases)? At the time of this writing, the latest pre-release is 3.3.6 and release 3.3.0. Though, only a JARs are provided and no DMG. At BFH, we are usually providing our students binaries built from the development HEAD and had no major issues with them.

John-Lluch, 2020-09-21:
Hi @maehne, I'm sorry, but I still get version 2.14.6 when I download it from the link in the readme file. I've just done it again a few minutes ago. So, please can you help me to find the location of the latest version. That does not look any obvious to me. Thanks in advance

John-Lluch, 2020-09-21:
Ok, never mind. I found "logisim-evolution-3.3.0-all.jar" in the releases section. However when I attempt to launch it on my computer a pop up appears with the following message
`The Java JAR file "logisim-evolution-3.3.0-all.jar" could not be launched. Check the Console for possible error messages`
but I can't find any related messages on the Console app. I'm using Mac OS 10.12.6 with "Java Version 8 Update 221". Any help is appreciated.

maehne, 2020-09-21:
Logisim-evolution requires since some time now at least OpenJDK 9 or above. I know that OpenJDK 14 works well (didn't find time yet to test with OpenJDK 15, but I would expect it to work). Also the OpenJDK 11 LTS release is an option. Though, there might be some issues, which got fixed in more recent JDK releases (e.g., regarding High DPI screens).

maehne, 2020-09-21:
I would recommend version 3.3.6 over 3.3.0 because it fixes some known bugs. Even the pre-releases are quite stable due to the mature code base.

maehne, 2021-01-27:
@John-Lluch: We now propose [installers for different platforms](https://github.com/reds-heig/logisim-evolution/releases), which are generated with `jpackage` from OpenJDK 15. You may want to give them a try. They also integrate a minimal JRE.

closed 2021-07-14 by BFH-ktt1