logisim-evolution/logisim-evolution #441: Command Line Verification Bugs

mfbutner, 2020-09-27:
I believe that something is broken with the command line verification with both the 3.3.0 release as well as the 3.3.1 release. 

## 3.3.0
For 3.3.0 I downloaded the prebuilt jar and am running it on Windows with Java 15. When I run the jar by double-clicking it or from the command line everything is fine. But when I add `-tty table` to the command line I get the following error
[image] and the program closes. This problem is fixed in 3.3.1 but I thought I should bring it up as 3.3.0 is the current advertised release.

## 3.3.1
I tried downloading and building the current code on the master branch on my Linux machine that has Java 11.0.8 on it. Again, everything is working fine until I attempt to add `-tty table` to the command-line options. This time however the program does not crash but only outputs one line of logging when there should be a lot more. It turns out the reason for this error was that I had a **tunnel** named halt in addition to the pin named halt. The documentation only mentions the halt pin and nothing about tunnels so I was very surprised by this behavior.

maehne, 2020-09-27:
@mfbutner: Thanks for reporting this bug! Could you please provide a small reproducer for the issue? We are not actively using the command line verification feature in our courses. Therefore, it is indeed possible that something broke without us noticing.

mfbutner, 2020-09-28:
Sure. Here are two examples. 
- The one with tunnel named stop works: [command_line_example_tunnel_named_stop.circ](https://drive.google.com/file/d/19Eon35a2ExhZzsIuRYgznGHVfTYz5gk0/view?usp=sharing)
  - `java -jar path_to_logisim_evolution command_line_example_tunnel_named_stop.circ -tty table` 
  - You should get something like 
  - 0       0       0       0
1       1       0       1
0       1       0       1
1       0       1       1
0       0       1       1
1       1       1       0
- This one with the tunnel named halt breaks.[command_line_example_tunnel_named_halt.circ]([command_line_example_tunnel_named_stop.circ])
  - `java -jar path_to_logisim_evolution command_line_example_tunnel_named_halt.circ -tty table` 
  - This will give you something like 
  - x a b A_XOR_B y
0 0 0       0 0

Let me know if there is another way you would prefer me to share the examples. 

I have experienced some other bugs with
- If a tunnel and another element have the same name moving the element with will erase its name
- If a tunnel and another element have the same name, if you save the file, close the file, and then reopen the file the elements with the same name as the tunnel will be erased. 
- If you load a circuit and change its appearance from classic to logisim-evolution, after you save the file, close the file, and then reopen it, it will revert back to the classic logisim appearance which will cause all of the lines to disconnect from the input/output port.

I'll post separate bug reports for these to help keep things organized.

maehne, 2020-09-29:
Thanks for the test cases. We will have a look. Though, it may take some time due to a busy term.

mfbutner, 2020-09-30:
Thanks for taking a look. I totally understand if it takes a while. Classes are starting tomorrow for us so it is really busy here too.

BFH-ktt1, 2021-04-07:
@mfbutner , please try with the latest version in the development tree, I think that the problem was there as I did not yet complete the merging of Kevin Walsh's chronogram code. I cannot reproduce the error anymore, hence I close the issue, please reopen if the problem still persists.

closed 2021-04-07 by BFH-ktt1