logisim-evolution/logisim-evolution #442: Tunnels and Components Not allowed to share Labels

mfbutner, 2020-09-28:
If a tunnel and another component share the same label (have the same name) the following errors occur
1. When you move the element with the same label as the tunnel, the element loses its label
2. If you have a tunnel with the same name as an element, save the file, close the file, and then open it up again the elements with the same name as the tunnel will lose their name. 

I'm not sure if this is intentional so as to keep everything having unique names but it contradicts a lot of what I did with original logisim where I would have say a tunnel and output/input pin with the same name.

maehne, 2020-09-29:
This behaviour is as far as I know intentional (@BFH-ktt1: you may correct me), as unique identifiers are required for enabling netlisting the design to VHDL or Verilog using the FPGA commander. Pin names have priority over gate/sub-circuit instance names, as the latter can be assigned automatically and the former are part of a sub-circuits interface definition. Is the current silent fixing behaviour bothering you? Would you rather like to have a warning?

mfbutner, 2020-09-30:
With all of the components aside from tunnels a warning already pops up saying you can't name two things the same thing if you try to.

If possible I'd like to have tunnels be named the same thing as other components because tunnels just connect wires and aren't really components.  I also like being able to name the tunnel that connects to pin X, X as they have the same value. If that isn't possible I would also like to get a warning saying I can't do it.  I wasted a lot of time putting the labels on again after they disappeared because I kept thinking I had forgotten to add them.

BFH-ktt1, 2021-04-07:
@mfbutner , Also this issue should be solved in the latest version, please reopen the issue if it is not the case

closed 2021-04-07 by BFH-ktt1

mfbutner, 2021-04-08:
Fantastic :)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>