logisim-evolution/logisim-evolution #444: [Windows 7 64 bits] Font is way bigger and bold and can't seem to be changed?

leandor, 2020-10-12:
Hello, 

Unless I'm blind... sorry if that's the case, looked everywhere in all the settings windows, and didn't find it.

I was using the original Logisim and just now found this version and was really exited to try it. I've installed the recommended Java version and running the program through it and after some hiccups managed to get it to run.

Everything seems to work fine, except that for some reason, in this version everything looks horribly bigger and out of proportion, and I'm not sure if that's how it's supposed to look, or if it's some quirk of this Java VM on Windows that perhaps it's biting me. I'm not used to run Java programs, and I'm not familiar with troubleshooting Java issues myself.

Is there maybe a hidden way to tweak the appearance, maybe change fonts? Decrease size and remove the boldness? (I hate bold fonts, aesthetically... and that seems to be the biggest offender and the cause of the rest of alignment issues.)

And what I mean is that when I loaded my old circuit from the original version everything is way out of shape and in some parts stuff overlaps with itself due to bigger font and I wouldn't want to have to realign and reposition everything again.

In case it's of significance, running on:
```
openjdk version "11.0.8" 2020-07-14
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.8+10)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.8+10, mixed mode)
```

Also, version I was using is the latest available as of now:
```
logisim-evolution-3.3.6-all.jar
```

maehne, 2020-10-17:
Try to go to open `Window` -> `Preferences`. On the tabs `Window` and `Layout`, there are the most important settings regarding the look of the user interface. In particular, you'll find on the `Window` tab the settings `Zoom factor` and `Look and Feel`. Try to set the zoom factor to `1.0` and the look and feel to `Metal` and restart Logisim-evolution. This hopefully fixes your issue. I am not aware of a setting allowing you to change the fonts though.

leandor, 2020-10-17:
Thank you! Didn't think of that Zoom setting when I was looking at the preferences page.

Still it didn't help with my problem, sadly. Thing is, the Fonts all seem to be converted to Sans Serif 16 BOLD when opening a saved circuit file (i.e. in the conversion from the old Logisim) and that mess up all the text in the circuit which leads to stuff overlapping each other.

Here's a snapshot comparison side by side (first one is original Logisim, note the font Sans Serif 12 *plain*):
[image]

Had to make Logisim Ev window bigger to be able to show the properties details:
[image]

As you can see, the Preferences were changed in the way you suggested:
[image]

Thing is, even if I wanted to change the fonts one by one for each element, seems that I can't get rid of the *bold* aspect of them. Whatever font I chose, even if I change it to over font face and select plain, it always comes back as bold.

Hope this shows the problem a bit more in detail!

Thanks again for your reply!

MarcinOrlowski, 2021-07-13:
I am closing this one as we have separate tickets to track related issues already.

closed 2021-07-13 by MarcinOrlowski