logisim-evolution/logisim-evolution #445: Logisim Crashes while opening a file that contains an empty text box (fully debugged)

dayeol, 2020-10-31:
Hi, 

We are using Logisim for a lecture, and a group of students was having a problem where Logisim crashes whenever they open their circuit.
I was able to reproduce it and figured out the cause.

It happens if you saved your circuit while you were entering an empty string in a text box as follows:

[image]

This creates an invalid entry of component like the following:

```
<comp lib="8" loc="(669,925)" name="Text">
    <a name="font" val="SansSerif plain 12"/>
</comp>
```

Compared to what it should look like:
```
<comp lib="8" loc="(669,925)" name="Text">
    <a name="font" val="SansSerif plain 12"/>
   <a name="text" val=""/>
</comp>
```

And when you open this file, Logisim just hangs.

So the component is missing `<a name="text" val=""/>` only when you saved the file *while* you were editing the text box.
This doesn't happen when you saved the file after you put an empty string and got out of the text box.

I believe this is an unintended behavior.
I haven't programmed in Java so just decided to report the issue instead of contributing.

Thank you!

BFH-ktt1, 2020-11-03:
Thank you for your detailed bug report. I look into the details as soon as I find some time. This might take a while as the COVID-situation takes much of my "free time".

BFH-ktt1, 2020-12-23:
This should be fixed in my latest push. Actually, a textbox with an empty contents should not be saved at all, as you cannot see it anymore. The current solution is that all textboxes with empty contents are not saved any more, and hence will not be present when loading the file. If the file is not closed, they will still be present in the circuit as "empty"; modifying them is still possible, but "finding" them is kind of a hassle.

commit 89ec2a2c (2020-12-23): Fix #445
closed 2020-12-23 by BFH-ktt1