logisim-evolution/logisim-evolution #446: Cannot Load Memory Image Files

mfbutner, 2020-10-31:
I am unable to load memory image files to either RAM or ROM. This seems independent of the format that the file is saved in. I tried
- v2.0 raw
- v3.0 hex words addressed
- v3.0 hex words plain

I know it is not a formatting issue with the files because I would edit a memory file (which works) and save its contents to a file (which also works) but when I would go back to load the image I just saved, nothing happens (the contents of memory don't change).

BFH-ktt1, 2020-11-03:
Thank you for your bug report. I look into the details as soon as I find some time. This might take a while as the COVID-situation takes much of my "free time".

BFH-ktt1, 2020-11-19:
Dear @mfbutner , I could not reproduce your bug on the latest version (3.3.6). Could you please give that one a try and post some test-files in case you can observe the same problem. I close the issue for now, please reopen if the problem still persists in v3.3.6.

closed 2020-11-19 by BFH-ktt1

mfbutner, 2020-11-20:
3.3.6 Is working for me too :)