logisim-evolution/logisim-evolution #447: Add Topics

TheEvilSkeleton, 2020-11-13:
Hello, `logisim-evolution` doesn't show up in the `Logisim` topic: https://github.com/topics/logisim. I just wanted to let you know that more people will find this software when topics get added.

Here are more Topics that I recommend to add: `education` `circuit` `circuits` `digital-circuit` `digital-circuits` `simulator` `logic`

MarcinOrlowski, 2021-06-10:
Indeed. But that can be easily addressed by editing repo topics: https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/classifying-your-repository-with-topics

@maehne

BFH-ktt1, 2021-06-10:
I added the topics. When we require some others, please let me know. I'm closing the issue, please reopen if you think that the issue is not resolved

closed 2021-06-10 by BFH-ktt1

MarcinOrlowski, 2021-06-10:
I think you want to add `logisim` topic too

maehne, 2021-06-11:
`logisim` was added as a topic.