logisim-evolution/logisim-evolution #448: Flatpak for Linux

TheEvilSkeleton, 2020-11-14:
Hello, I've created a fully working flatpak for logisim-evolution: https://github.com/TheEvilSkeleton/com.github.reds.LogisimEvolution

I wouldn't mind if the project gets forked gets published it on Flathub.

BFH-ktt1, 2020-11-19:
Thanks a lot for your work. I will look into the details the moment my current packed schedule shows some room.

TheEvilSkeleton, 2020-11-19:
I completely forgot to update: I started using it. I used the flatpak for a whole day and it seems to be working just fine.

TheEvilSkeleton, 2020-11-26:
@BFH-ktt1 so in https://github.com/TheEvilSkeleton/Logisim-flatpaks/commit/c8baa8ccc0563510624904b4e13484de90d28fec, I made it show up in the "Open with" dialog:

[image]

The problem is that when the file name has a space, e.g. `~/Documents/Logisim/Logisim test.circ`, or the previous folder's name has a space, e.g. `~/Documents/Logisim test/Logisim.circ`, the file won't launch at all. I'm pretty sure it's an upstream issue, because I can launch it successfully when there is no space.

TheEvilSkeleton, 2020-11-26:
> Thanks a lot for your work. I will look into the details the moment my current packed schedule shows some room.

I wouldn't mind being one of the maintainers when it gets published in Flathub. Just that I'd like to collaborate with others so I don't have to do 100% of the work.

BFH-ktt1, 2020-11-26:
@TheEvilSkeleton , Concerning the spaces; this is a general problem a lot of unix-based software has, as in unix the space is used for separation between the program and it's argument(s). So in general it is a bad idea to use spaces in either the file-name or the directory name. If those spaces are used, for example in the workspace name, than also the syntehsize and download functionality will give problems. Of course this can be "fixed" by putting the filename in double quotes or by backschlassing the spaces, but this "fix" does not work for the software that is used by logisim to do the external simulation (modelsim/questasim) and/or the synthesis and p&r (Intel Quartus/Xilinx/Vivado). Therefore the best solution is to not use spaces in either filenames or directory names.

Concerning maintaining on flathub, thanks for your offer, and no, you will not be alone.

TheEvilSkeleton, 2020-11-26:
> Concerning maintaining on flathub, thanks for your offer, and no, you will not be alone.

Awesome. I'm going to submit an MR soon in Flathub.

TheEvilSkeleton, 2020-11-26:
@BFH-ktt1 https://github.com/flathub/flathub/pull/1951

TheEvilSkeleton, 2020-11-26:
> but this "fix" does not work for the software that is used by logisim to do the external simulation (modelsim/questasim) and/or the synthesis and p&r (Intel Quartus/Xilinx/Vivado). Therefore the best solution is to not use spaces in either filenames or directory names.

That's understandable, but also problematic, as the program doesn't support this basic feature. People will often think that the flatpak will broken when opening a file containing a space, when in reality it's an upstream issue.

I'm able to open files that have spaces from the program itself, but from the "Open With" dialog I cannot.

closed 2020-12-14 by TheEvilSkeleton

BFH-ktt1, 2020-12-14:
@TheEvilSkeleton , sorry for not being there for a longer time, but the last weeks of the semester are "killing" me. Will look in the open issues from next week on. Thanks for all your work.

TheEvilSkeleton, 2020-12-14:
@BFH-ktt1 don't worry, take your time, and I'm glad to help. I closed this issue because it is resolved.

Here's a recap if you were out of touch with what has happened:

- logisim-evolution is now available in the Flathub repository: https://flathub.org/apps/details/com.github.reds.LogisimEvolution, so it will show in GUI software centers like GNOME Software[1].
- The repository for the package is here: https://github.com/flathub/com.github.reds.LogisimEvolution. You should have an invitation to maintain the repository.
- I added some stuff like "Open with Logisim Evolution" to make opening `.circ` files easier[2].
- An icon should appear in your application launcher[3] (but will prefer to use an official icon).

[1]
[image]
[2]
[image]
[3]
[image]

BFH-ktt1, 2020-12-22:
@TheEvilSkeleton , I seem to remember that there was still a question open concerning flatpack files that should be "moved" to the logisim tree. I seem have lost this email, could you give me a pointer to the issue? Thanks in advance. I plan to make a V3.4.0 (which is the current development tree with some bug fixes) in the coming days.

TheEvilSkeleton, 2020-12-22:
@BFH-ktt1 https://github.com/flathub/flathub/pull/1951#discussion_r533142094, and https://github.com/flathub/flathub/pull/1951#discussion_r533142138.

BFH-ktt1, 2020-12-23:
@TheEvilSkeleton , I added the files in the logisim repository, and build a new stable version. It would be nice to have a small README on how to update flatpak wenn there is a new release, I'm completely new in building flatpak applications, so could you enlighten me?

TheEvilSkeleton, 2020-12-23:
@BFH-ktt1 it's really easy to bump versions. It's also convenient since the [Flathubbot](https://github.com/flathubbot) does the "building" process.

All you have to do is go to the manifest, and change the `url` and `sha256` lines (https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.yml#L50-L52); you type the URL you want to fetch, and then you type the hash of what you want to fetch. Once done, you can test if the flatpak works by submitting a merge request in the logisim-evolution repository. If it fails, then Flathubbot will comment it [failed](https://github.com/flathub/flathub/pull/1951#issuecomment-734214923), if it successfully builds, it will comment saying it's [successful](https://github.com/flathub/flathub/pull/1951#issuecomment-735945567).

Since you moved [some files](https://github.com/reds-heig/logisim-evolution/tree/master/support/Flatpak) to upstream, we will need to make the flatpak fetch those files instead of the [root repository](https://github.com/flathub/com.github.reds.LogisimEvolution). You can see the lines for the [`.desktop`](https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.yml#L42-43) and [MIME](https://github.com/flathub/com.github.reds.LogisimEvolution/blob/master/com.github.reds.LogisimEvolution.yml#L48-49), just a simple:

(Pardon for incorrect indentation. I'm using my phone right now)

```diff
-   path: com.github.reds.LogisimEvolution.desktop
+  url: URL
+  sha256: HASH
```

`dest-filename` is just renaming the file to something else. Since the [Flatpak directory](https://github.com/reds-heig/logisim-evolution/tree/master/support/Flatpak) has the right name, I don't think we have to rename.

You can give it a try. Just submit an MR in the [com.github.reds.LogisimEvolution](https://github.com/flathub/com.github.reds.LogisimEvolution) repository. I'll request some changes if you ever need some help.