logisim-evolution/logisim-evolution #449: Create a logo for logisim-evolution

TheEvilSkeleton, 2020-11-15:
As the title suggests.

maehne, 2021-01-10:
Have you checked out the icons, which are included in the new installer packages created with `gradle jpackage`? If you would like to suggest a new or enhanced logo to distinguish Logisim-evolution from the original Logisim and other forks. Please feel free to make a proposal and pull request!

maehne, 2021-01-27:
@TheEvilSkeleton: Are you satisfied by the logos used by the new installer packages for the program itself and Logisim circuit files? Can we close this issue or will you provide a PR with a proposal for a new one?

TheEvilSkeleton, 2021-01-27:
> @TheEvilSkeleton: Are you satisfied by the logos used by the new installer packages for the program itself and Logisim circuit files? Can we close this issue or will you provide a PR with a proposal for a new one?

Can you show me how it looks? I haven't really had the time to check on the logo.

maehne, 2021-01-27:
@TheEvilSkeleton, it looks like this:

[image]

So, no real difference to the icon of the original Logisim.

TheEvilSkeleton, 2021-01-27:
I guess we could close the issue since it looks fine. I'd like to come up with a better logo but I don't really have the time.

closed 2021-01-27 by TheEvilSkeleton