logisim-evolution/logisim-evolution #451: Logisim-evolution don't keep saved custom circuit appearance when circuit is loaded as library.

nahinulsadad, 2020-11-22:
Logisim-evolution version: 3.36
OS: Windows 10
Java Version: 13.0.2 (JRE build 13.0.2+8)

I have saved the circuit appearance of FA when I opened it as file. Logisim-Evolution works just fine and keeps circuit appearance but it do not save facing of the circuit.

[image]

But when I load it as library in Logisim-Evolution, It don't keep correct circuit appearance and revert back to original appearance.

[image]

Even though other circuits in that library uses its correct appearance.

[image]

I tried the same library (created in Logisim-Evolution) in Logisim-original and it works just fine.

Logisim-Evolution File:
[Clean_Multiplier_Library.zip](https://github.com/reds-heig/logisim-evolution/files/5578892/Clean_Multiplier_Library.zip)

BFH-ktt1, 2020-11-22:
Dear @nahinulsadad , thank you for your report. This issue is a kind of duplicate of #334 (I did not yet observe that the appearance inside the library is preserved, which gives me more info, thank you for that detail) and I will look into the details as soon as I have again some time available.

BFH-ktt1, 2020-12-23:
Is fixed in v3.4.0. Closing the issue, please reopen if the problem reappears

closed 2020-12-23 by BFH-ktt1