logisim-evolution/logisim-evolution #452: [Windows 10] "This app can't run on your PC"

TheEvilSkeleton, 2020-12-01:
Hello, I'm trying to directly open a `.circ` file in Windows 10 but it gives me this error:

[image]

I do not know how to troubleshoot manually, so I will need guidance.

BFH-ktt1, 2020-12-02:
You have to associate the files with a ".circ" extension to logisim. As I am not a windows user, I do not exactly know how to do that. By default windows wants to "execute" the .circ file with which it fails.

TheEvilSkeleton, 2020-12-03:
That's what I did; I tried to set logisim-evolution as the default app for the `.circ` file type:

[image]

Opening the `.circ` file brings to this issue: https://github.com/reds-heig/logisim-evolution/issues/452#issue-754544568

wgaylord, 2020-12-11:
It tried to just run the .jar which windows can't do as you need to invoke java -jar then the jar file name.

TheEvilSkeleton, 2020-12-11:
I'd personally appreciate a simple double click.

wgaylord, 2020-12-12:
Do you have ".jar"  files associated to Java? if not try that and it might fix it.

TheEvilSkeleton, 2020-12-16:
Yes, I'm using [AdoptJDK](https://adoptopenjdk.net/) for this and logisim-evolution starts as it should. The only problem is that double clicking on `.circ` files can only open in Logisim, but it cannot be opened in logisim-evolution. [I tried to set logisim-evolution as the default, but it doesn't work.](https://github.com/reds-heig/logisim-evolution/issues/452#issuecomment-737607998)

maehne, 2020-12-17:
It is not possible to directly associate `.circ` with a `.jar`-Archive, as this is not a natively executable format on Windows. However, you can achieve it by launching Logisim-evolution using a batch/shell wrapper script. Unfortunately, you'll have to build Logisim-evolution from source using `./gradlew shadowDistZip` in the root of the cloned Logisim-evolution Git repository. This will build Logisim-evolution and create a distribution archive `./build/distributions/logisim-evolution-3.?.?.zip`. This archive can be unpacked on Windows into a global folder. (I personally use `C:\Apps\` for manually installed software.) The folder `logisim-evolution-3.?.?\lib` contains the `.jar` archive and the folder `logisim-evolution-3.?.?\bin` the desired batch wrapper `logisim-evolution.bat`. Right-clicking on it, you can create a shortcut to that batch file, which you can move to your Windows start menu. Then, you can associate `.circ` files with Logisim-evolution.

I agree that this is currently a cumbersome manual process, which should get automated (and it was actually until Oracle decided to remove without timely replacement `jarpackager`). With the advent of `jpackage` in OpenJDK14, it might be possible again to generate a nice installer for different platforms. Unfortunately, I currently don't find the time to work on this issue due to a very high teaching work load.

BFH-ktt1, 2020-12-25:
This should now be resolved by V3.4.1, closing the issue

closed 2020-12-25 by BFH-ktt1