logisim-evolution/logisim-evolution #454: Board mapping: Hidden pop-up under Windows (issue for version 3.3.6)

igd1, 2020-12-02:
Mapping in-/outputs will trigger, dependent on the selection, a pop-up that freezes all other Logisim windows. Using Windows 10, this popup is hided behind the board mapping window. There are two solutions: 
- Switch the version --> no issue for version 3.3.4 and earlier.
- You can access the pop-up when opening the task tanager and minimizing the Logisim sub process "Synthesize & Download:..."

maehne, 2020-12-04:
I can confirm this issue. Another workaround is to use `ALT+SPACE` to access the window menu and then move the window hiding the pop-up.

BFH-ktt1, 2020-12-22:
My latest push should fix this issue.
@maehne , it should also fix the bug we discussed this morning that is not yet marked as issue.
Could you please test and let me know if I can close the issue?

BFH-ktt1, 2020-12-23:
Fixed in v3.4.0, closing issue, please reopen if not resolved

closed 2020-12-23 by BFH-ktt1