logisim-evolution/logisim-evolution #461: Infinite loop when loading 2.x circuit file in 3.x

zdimension, 2020-12-23:
(tldr at bottom) I've tried opening a few circuits I made on Logisim-evo 2 in Logisim-evo 3 and it freezes while reading the file. Upon further inspection, the problem comes from XmlCircuitReader.java, line 250~254
```java
Bounds bds = comp.getBounds();
int d = 0;
do {
  d += 10;
} while (componentsAt.get(bds.translate(d, d)) != null);
```
There seems to be cases where bds is (0, 0), and bds.translate(d, d) returns (0, 0) too, so the loop ends up checking the same condition every time, freezing the program.

I'm not sure what this line in Bounds.java is supposed to do:
```java
if (this == EMPTY_BOUNDS) return this;
```
If Bounds is supposed to represent a 2D rectangle, then why would translating the empty rectangle give the same rectangle? It should still change the x and y values, or maybe I've missed something.
By looking at the contents of `comp`, I found that it corresponds to this line in my file:
```xml
<comp lib="8" loc="(485,179)" name="Text"/>
```
An empty Text component, which must have been placed there by mistake.

**tl;dr:** empty Text components in circuits cause infinite loops in 3.x

BFH-ktt1, 2020-12-23:
@zdimension , thanks a lot for the detailed bug report! I'll look into the details. It is related to an issue where logisim saved "empty" text boxes (hence a text with zero contents). This is fixed, as v3.4.0 will not any more save empty text-boxes, however, an older version might and should not break V3.4.0, which it does in the moment.

BFH-ktt1, 2020-12-23:
Please check, this should resolve the problem

zdimension, 2020-12-23:
I confirm the problem is solved (at least for the file I'm using). Thanks again for the quick feedback!

BFH-ktt1, 2020-12-23:
Your welcome, closing the issue.

closed 2020-12-23 by BFH-ktt1