logisim-evolution/logisim-evolution #475: Download logisim-evolution on Mac (mac os big sur)

Jo-wang, 2021-01-17:
Hi, after I download and install it in application folder, it shows 
"logisim-evolution” is damaged and can’t be opened. You should move it to the Bin."
Is there anyone know why?
Thanks.

closed 2021-01-17 by Jo-wang

maehne, 2021-01-17:
Could you please give a reason why your opened and then nearly instantly closed this issue? On our side, the created app is working. However, I have not yet updated to Big Sur.

Jo-wang, 2021-01-17:
Because I forgot to unlock the security requirement on Mac. And now it’s totally can be run on Mac OS big sur.

maehne, 2021-01-18:
Thanks for this feedback @Jo-wang! Good to know that Logisim-evolution is also working on macOS Big Sur!

HongxiaoN, 2021-01-20:
Sorry, may u teach me how to unlock the security requirement? I try to give Logisim-evolution all the access, and it still does not work for my macOS Big Sur. @Jo-wang @maehne