logisim-evolution/logisim-evolution #477: Not launching

maniek207, 2021-01-21:
Debian 10 with openjdk 11
Tried .jar and .deb installer.
```
Exception in thread "main" java.lang.ExceptionInInitializerError
	at com.cburch.logisim.prefs.AppPreferences$LocalePreference.<init>(AppPreferences.java:90)
	at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:497)
	at com.cburch.logisim.Main.main(Main.java:53)
Caused by: java.lang.IllegalArgumentException: Illegal pattern character 'r'
	at java.base/java.text.SimpleDateFormat.compile(Unknown Source)
	at java.base/java.text.SimpleDateFormat.initialize(Unknown Source)
	at java.base/java.text.SimpleDateFormat.<init>(Unknown Source)
	at java.base/java.text.SimpleDateFormat.<init>(Unknown Source)
	at com.cburch.logisim.util.LocaleManager.<clinit>(LocaleManager.java:252)
	... 3 more
Failed to launch JVM```

maehne, 2021-01-25:
The most recent release of Logisim-evolution got built using OpenJDK 15 due we use its `jpackage` tool to generate the binary installers on the different supported platforms. Therefore, OpenJDK 11 is probably to old to execute Logisim-evolution. Could you please try installing OpenJDK 15 and execute `logisim-evolution` again?

GamePlayer-PL, 2021-01-26:
I will join the discussion,

So in OpenJDK 11: yes, it doesn't work.

I tried to run it on OpenJDK 15, downloaded from https://jdk.java.net/15/, but it produces the same error.

I also tried to recompile and run it on IntelliJ but again the problem is the same.
No error occurred during recompilation.

Does Logisim Evolution need any additional libraries to be installed than just OpenJDK 15?

BFH-ktt1, 2021-01-26:
Please try following:
rm ~/.java/.userPrefs/com/cburch/logisim/prefs.xml
and start logisim.
Maybe there is a setting from another version of logisim on which the v3.4.1 fail to start-up (such for example as a non-supported language)

GamePlayer-PL, 2021-01-26:
When I deleted prefs.xml file and reopened the jar file, I logged in with:

>   sty 26, 2021 10:32:45 AM java.util.prefs.FileSystemPreferences$6 run
>   WARNING: Prefs file removed in background /home/gameplayer/.java/.userPrefs/com/cburch/logisim/prefs.xml
>   Exception in thread "main" java.lang.ExceptionInInitializerError
>   	at com.cburch.logisim.prefs.AppPreferences$LocalePreference.<init>(AppPreferences.java:90)
>   	at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:497)
>   	at com.cburch.logisim.Main.main(Main.java:53)
>   Caused by: java.lang.IllegalArgumentException: Illegal pattern character 'r'
>   	at java.base/java.text.SimpleDateFormat.compile(SimpleDateFormat.java:846)
>   	at java.base/java.text.SimpleDateFormat.initialize(SimpleDateFormat.java:654)
>   	at java.base/java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:625)
>   	at java.base/java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:600)
>  	at com.cburch.logisim.util.LocaleManager.<clinit>(LocaleManager.java:252)
>   	... 3 more

maehne, 2021-01-26:
I found a few minutes to test building of the `master` branch on Debian Buster with OpenJDK 11 and `gradle`. It works well as long as you don't use the `jpackage`. Note, you should use the "shadow" tasks, which will include all library dependencies inside the JAR. I would recommend:

```sh
cd logisim-evolution/
./gradlew shadowDistTar
cd /usr/local/
sudo tar xfv <your-path-to>/logisim-evolution/build/distributions/logisim-evolution-shadow-3.4.1.tar
```

The task `shadowDistTar` creates a distribution TAR archive with the JAR plus a shell wrapper. The last two commands unpack that archive to `/usr/local/`. The shell wrapper ends up in `/usr/local/bin/` and thus in your `PATH` (depending on your shell, you might need to issue a `rehash` so that it is found) and the JAR in `usr/local/lib/`.

If you build Logisim-evolution using OpenJDK 15, you can create the DEB package using `./gradlew jpackage`. After building finished, you should find the DEB package in `build/dist/`.

maniek207, 2021-01-26:
I tried using OpenJDK 15 but still same error. Also on old laptop with Debian where this same issue was, I tried change system language from Polish to English and it's launched without problem. And on normal version on logisim works normal even with Polish.

GamePlayer-PL, 2021-01-26:
About idea of compiling it:

> ```shell
> cd logisim-evolution/
> ./gradlew shadowDistTar
> cd /usr/local/
> sudo tar xfv <your-path-to>/logisim-evolution/build/distributions/logisim-evolution-shadow-3.4.1.tar

Well, still it doesn't work.

I tried the idea of ​​@maniek207 about changing OS language. I tested it on a virtual machine and the program starts normally.

So the problem concerns languages, and more precisely the Polish language.

maehne, 2021-01-26:
Thanks @maniek207  and @GamePlayer-PL for the feedback. I can now reproduce the bug:

```
sudo dpkg-reconfigure locales
# Add pl_PL.UTF-8 as additional locale
export LANG=pl_PL.UTF-8
logisim-evolution
```

Then, Logisim-evolution gives me exactly the reported exception. So, the error is probably located in one of the `*.properties` files below `src/main/resources/resources/logisim/pl/`.

maehne, 2021-01-26:
I checked the properties files and of cause found the bug in the very last one (`util.properties`) on the check list. The `dateFormat` string was for some reason wrong. This is fixed by pull request #479.

@maniek207 and @GamePlayer-PL: Could you please confirm that [my branch `fix-issue-477`](https://github.com/maehne/logisim-evolution/tree/fix-issue-477) also corrects the problem on your side?

GamePlayer-PL, 2021-01-26:
On my side, the problem is gone. Logisim now starts up without a problem. :+1:

maniek207, 2021-01-26:
Now it's launch without any problems at my side.

maehne, 2021-01-27:
@BFH-ktt1: This issue can be closed, as it is resolved.

closed 2021-01-27 by BFH-ktt1