logisim-evolution/logisim-evolution #478: [Linux Mint 20.1] Problem with loading data to register element

GamePlayer-PL, 2021-01-25:
Hello :wave:,

A few minutes ago I ran into a binary registry problem in my test project (code: https://gist.github.com/GamePlayer-PL/b1cee86fb60af9639a6d2ab5f7118a68).

I think exactly this error giving an invisible signal to register. I tried to show it in this video as well.

https://user-images.githubusercontent.com/57105419/105753580-3b797900-5f49-11eb-8887-8876684b161d.mp4

Before writing this issue, I checked this problem on other computers and it keeps repeating. I also checked gates in the project and everything should be ok.

Honestly I don't know how to fix this and sorry if my english is bad.

Specific informations:
 - Logisim version: From apt package list - 2.7.1 (version 3.4.1 not working - i will make next issue about it)
 - Tick freq.: 1 Hz
 - OS: Linux Mint 20.1 Cinnamon
 - Linux Kernel: 5.4.0-64-generic
 - Cinnamon version: 4.8.6

maehne, 2021-01-25:
Your design is not fully synchronous because you are generating individual clock signals via combinatoric circuits, which generate hazards, which very likely cause the undesired behaviour you are observing in your circuit. When designing sequential circuits, I recommend you to use always a single clock signal, which is directly connected to all D flip-flop clock inputs without passing through any combinatoric gates.

I would not recommend you to use the ancient Logisim 2.7.1 version as it is not maintained anymore by the original developer Carl C. Burch. The most recent version of Logisim-evolution works well on Linux. Though, the DEB package got generated with `jpackage` from OpenJDK 15, which probably explains your trouble on Debian Buster with old OpenJDK 2011, which you reported in issue #477.

GamePlayer-PL, 2021-01-26:
Ok, so next time I should use less gates to generate one signal. 

But if this problem be minimized in the future or is it rather impossible?

maehne, 2021-01-26:
No, it is not a question of how many gates you use to create one signal. Using gated signals als clocks is always problematic due to hazards and addition of skew and therefore not encouraged for digital designs. The distribution of clock signals requires careful routing. Especially FPGAs only provide a limited number of clock and reset trees for distributing these critical signals. Therefore, clock enabling using a multiplexer in front of the D input of the D flip-flops, which forwards either the current value Q of the flip-flop or the next value from the next state logic into the D input based on the clock enable signal being inactive or active, respectively, is the recommended approach. For details, see, e.g., Section "Clock Enable" on page 26 of the [Lattice HDL Coding Guidelines](http://www.latticesemi.com/view_document?document_id=48203).

For reliable operation, you will therefore need to modify your design. Any measures inside Logisim to ignore hazards would reduce the fidelity of its simulation with respect to real hardware.

GamePlayer-PL, 2021-01-26:
so if one signal has more gates than the other, the signal will have a slightly longer delay than other signal. Ok, now I know, what the problem is. :+1: 

I have some suggestion: 
Would there be an option to add a history of objects? More precisely, what happened to a given object at a certain time. This would allow to analyze the circuit in at a certain time and possibly return to the appropriate point in the object or circuit.

History would not be recorded to file, and it could remember the postural movements of an object for some time.

Yeah, it can eat much more memory but the given activity record of object would be removed from history after 5/10 minutes, for example.

maehne, 2021-01-26:
The longer delays caused by the gates are only a part of the problem. Worse are the temporal signal instabilities caused by them each time an input of a combinatoric sub-circuit changes its values. These are propagated along several paths. Due to different path delays the sub-circuit outputs can change one or more times until it stabilises. These disturbing impulses, called "hazards", may or may not trigger one or more unintended sampling(s) of the D inputs of D flip-flops.

The feature you are asking for, if I understand you correctly, is essentially the already implemented "Chronogram" feature. You can find usage instructions in the program's help system by searching for "chronogram". I agree that it is a bit hidden, only sparsely documented, and its usability could be improved, but we have only limited development resources at our disposal. Pull requests are welcome!

GamePlayer-PL, 2021-01-26:
> The feature you are asking for, if I understand you correctly, is essentially the already implemented "Chronogram" feature. You can find usage instructions in the program's help system by searching [...]

In older version I doesn't seen it. Now good to see it on newest version. :smile: 

> The longer delays caused by the gates are only a part of the problem. Worse are the temporal signal instabilities caused by them each time an input of a combinatoric sub-circuit changes its values. [...]

Yeah, that's right.

So, in order not to drill down on the topic, it's time to solve it.

**Solution**:

The strange behavior of the boundaries in the perimeter can apply to many things:
 - Old version of Logisim;
 - delays from gates and other objects;
 - Unevenly running clocks;
 - Incorrectly connected circuits;
 - computer and program limits.

Due to the fact that Logisim is to simulate the operation of the circuits, the delay will not be removed.

So not always strange behavior of circuit mean error or bug (one of strange behavior you can see on my example circuit).

So i'm closing issue.

closed 2021-01-26 by GamePlayer-PL

maehne, 2021-01-26:
Thanks @GamePlayer-PL for closing the issue. Don't hesitate to open new issues! Any feedback and contributions in form of code, documentation, ideas, etc. are welcome!