logisim-evolution/logisim-evolution #486: Logic gates graphical bug?

Von-Braun, 2021-01-27:
For some reason, my gates are appearing in box form, rather than as their respective symbols. Any idea on why this is? My OS is Windows 10 64-bit.
```
> java -version
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF8
java version "1.8.0_281"
Java(TM) SE Runtime Environment (build 1.8.0_281-b09)
Java HotSpot(TM) 64-Bit Server VM (build 25.281-b09, mixed mode)
```

[image]

BFH-ktt1, 2021-01-27:
@Von-Braun , You have on the settings page selected the IEC-Symbols instead of the ANSI-Symbols. You can change it back by
Window->Preferemces->International->Gate Shape->ANSI

Von-Braun, 2021-01-27:
@BFH-ktt1 Welp, now I feel like an idiot for spending over an hour on this, haha. Thanks very much!

closed 2021-01-27 by Von-Braun