logisim-evolution/logisim-evolution #487: Hex editor for RAM: unable to edit words wider than 32 bits

regus-pregus, 2021-01-28:
Software version: 3.4.1
Problem description: when RAM component data bit width parameter is selected to more than 32 bits, hex editor displays enough characters per word to fit specified number of bits (e.g. for 64-bit width it shows 16 hex characters per word, which is correct), however, editing the value truncates at 32 bits (4 bytes).
Repro:
0) Start logisim-evolution version 3.4.1
1) Place RAM component on canvas
2) Specify Data Bit Width = 64 in Properties window
3) Right-click on RAM component on canvas, select Edit Contents... menu item, hex editor will launch
4) Select any cell in hex editor and start entering values. Only first 8 characters will enter correctly, after which new characters will be displacing old characters instead of overflowing to the upper 8 characters. Upper 8 characters (4 bytes) will always remain zero.

maehne, 2021-01-28:
Thanks @regus-pregus for the bug report! I can reproduce the wrong behaviour. Interesting that it works for the first word, but not the other memory places. It's probably a relatively small cause overlooked, when the maximum bus width was increased from 32 bit to 64 bit. @BFH-ktt1: Can you have a look?

maehne, 2021-01-28:
@BFH-ktt1: In this [branch](https://github.com/mkayaalp/logisim-evolution/tree/unh-ece562-spring2020) by @mkayaalp might be a part of the solution to fix the issue.

BFH-ktt1, 2021-01-29:
@regus-pregus : It should be fixed by my latest PR. Please reopen if the problem still persists.

commit 48769c4d (2021-01-29): Fix #487
commit be2fb96c (2021-01-29): Merge pull request #489 from BFH-ktt1/chronogram

Fix #487
closed 2021-01-29 by BFH-ktt1