logisim-evolution/logisim-evolution #494: Upgrade to Java 11

zdimension, 2021-02-01:
The project is currently set to use Java version 9, whose support ended in March 2018. 

It would be better to switch to a more recent version, or at least to the next LTS version, Java 11.

maehne, 2021-02-02:
I agree. Historically, we always set the minimum version to the one, which would be still able to build without failure Logisim-evolution. We raised that version only when the integration of new features made that necessary. I think that modernising the code base and removing dependencies on deprecated APIs and libraries is a good thing and we therefore usually quickly accept PRs going in that direction.

Supporting also older JDKs makes it easier for people to build Logisim-evolution in case they are not regular Java developers and therefore would need to install manually the newest JDK just for that purpose. I think raising the minimum version to OpenJDK 11 is in this respect a good compromise, as it is the version, which should be at least available for installation through the package manager of most current Linux distributions (e.g., Debian Buster is stuck at OpenJDK 11).

Also, less people are forced to build Logisim-evolution on their own since we regained the possibility to create self-contained installers.

So, from my side feel free to create a PR in that sense. At the same time, it would be good to update [`.travis.yml`](https://github.com/reds-heig/logisim-evolution/blob/master/.travis.yml) to include also the more recent JDK versions (it currently stops at OpenJDK 13).

BFH-ktt1, 2021-02-02:
@zdimension , Can we close this issue as it was merged with your latest PR?

zdimension, 2021-02-02:
Yes, I'm closing it right now.

closed 2021-02-02 by zdimension

maehne, 2021-02-02:
I would suggest to add OpenJDK 14 an 15 to `.travis.yml`. Maybe, it would be even sensible to just test with OpenJDK 11 and OpenJDK 15, i.e., with oldest and newest supported JDK unless we get reports concerning JDK versions in between?

zdimension, 2021-02-02:
I agree with that. I'll change it and make a PR.