logisim-evolution/logisim-evolution #499: doesn't run on MacBook M1 with Big Sir

vladimirzhidkov, 2021-02-13:
constantly getting “logisim-evolution” is damaged and can’t be opened. You should move it to the Trash.

maehne, 2021-02-13:
Please check out issues #475 and #476 for solutions to your problem. Basically, you need to unlock the security requirement on macOS Big Sur. @davidhutchens: Do you have maybe an idea, how we can address this problem in the future so that Logisim-evolution runs out of the box?

davidhutchens, 2021-02-13:
I don't have an M1 processor Mac. I am also not running Big Sir. However, from what I see researching those issues, it should work under Apple's x86 emulator with the solutions @maehne noted above.

Once the M1 Mac is fully supported by the Java distributions, we should be able to compile it native to the M1.

As for fixing it so it works out of the box, I think we would need a developer account and code-sign it when building the distribution. I haven't tested that either.

vladimirzhidkov, 2021-02-13:
I can confirm that it works on MacBook M1 with Big Sur if you run a jar file directly from terminal.
java -jar /path/to/logisim-evolution-3.4.1-all.jar. Not a very elegant solution.
@maehne could you please teach me what exactly I need to unlock in the security on macOS Big Sur for it to run properly? 
I have tried rebooting and xattr -r /Applications/logisim-evolution.app. Neither helped
Thank you

vladimirzhidkov, 2021-02-13:
great app by the way. love it

davidhutchens, 2021-02-13:
On older systems, all you actually had to do was control-click on the app and select open from the menu. It would then give you the option of authorizing it to run (which requires an administrator's password). Once it was authorized, it ran normally from then on. You might need to set System Preferences - Security to allow identified developers as well as the App Store first. Try that and let me know if it still works.

vladimirzhidkov, 2021-02-13:
So after running xattr -cr /Applications/logisim-evolution.app (last time I forgot to add -c) in the terminal, the application starts flawlessly. So I guess xattr -cr /Applications/logisim-evolution.app is the solution to macOS Big Sur running on M1 chip.
thank you everyone for your help!

closed 2021-02-13 by vladimirzhidkov

ridicolos, 2022-11-08:
> Once the M1 Mac is fully supported by the Java distributions, we should be able to compile it native to the M1.

When will we see a native build for M1?

arkamn, 2022-11-09:
https://github.com/logisim-evolution/logisim-evolution/releases/download/v3.8.0/logisim-evolution-3.8.0.dmg