logisim-evolution/logisim-evolution #501: Order in Z axis is not saved

Aerrios, 2021-02-14:
Hello, 
when I save Circuit Appearance, order of shapes in "Z axis" is not saved.

Original design:
[image]

Design after loading circuit from file:
[image]

Emil-DV, 2021-06-01:
I noticed this as well and it is quite a headache since I'm going for a sunken 'battle ship' panel look.
I was able to set up the build environment and found that the issue was with a "sort" function that was added somewhere along the way for easier testing via .circ file compares (so say the comments in the code).  I commented out the call to sort (logisim-evolution-develop\src\main\java\com\cburch\logisim\file\XmlWriter.java line 197) and the Z order works now.

R3dst0ne, 2021-07-04:
Since #708 fixed this issue, can it be closed?

maehne, 2021-07-12:
Closing this issue. @Arri0: Feel free to reopen this issue, if it should persist.

closed 2021-07-12 by maehne