logisim-evolution/logisim-evolution #504: N-transistor behavior is broken

vladimirzhidkov, 2021-02-15:
here is a simple circuit with one transistor in the original Logisim:
[image]
it works properly

and this is the same circuit in logisim-evolution:
[image]
does not work properly

when N-transistor is replaced with P-transistor, the circuit is back to working properly

[image]

maehne, 2021-02-16:
According to the tool tips, the NMOS behaves correctly "a low source will pull drain low (only when gate is high)". Otherwise, drain will be high impedance or conflict. In your examples, you have connected the NMOS source to high.

Logisim is a digital simulator and therefore its transistor models are very simplified and restricted to illustrate the functioning of logic gate transistor circuits in common topologies, i.e., PMOS transistors are located above the output constituting a current path from VDD to the output and NMOS transistors are located below the output constituting a current path connecting the GND to the output. Other circuit topologies might not behave as expected due to the very simplified model.

vladimirzhidkov, 2021-02-17:
Excuse my incompetence, I am fairly new to logic circuits. I built the exact same buffer circuit in both the original Logisim and  the Evolution one. The results did not match (see the pictures above). So I concluded that one of them must behave incorrectly. I assumed it was Evolution since my buffer circuit did not work in it.

closed 2021-02-17 by vladimirzhidkov

maehne, 2021-02-17:
Indeed, the behavioural difference between Logisim and Logisim-evolution may hint that there's anyway an issue. @BFH-ktt1: Maybe, you could have a look, too?

In any case we appreciate all bug reports -- even if it turns sometimes out that it's not actually a bug! They, e.g., give us hints were documentation can be improved. This may be the case for this issue.